betaflight/betaflight #4580: Feature Request : One button to bind to Spektrum receiver.

davecraige, 2017-11-14:
Hey guys,

Thanks for all the great work on BetaFlight.  Appreciate all the countless hours that went into creating and improving it.

Request: One button to bind to Spektrum receiver
Would it be possible to simply set up a button on one of the pages where you could click to begin a bind with a Spectrum receiver.  This would help many new users (like myself) to have to use the CLI and run the "set spektrum_sat_bind = 9" code.  

Thoughts?

jflyper, 2017-11-14:
There must be something other than a button to specifiy the bind code.

https://github.com/cleanflight/cleanflight/blob/master/docs/Spektrum%20bind.md#table-with-spektrum_sat_bind-parameter-value

mikeller, 2017-11-15:
@jflyper: The problem is that the bind signal has to be sent within a short timeframe after the RX is powered up. Doing so directly only really works with a button, for everything else the activation needs to be persisted, and then executed after the next hard reset.
But adding an MSP command to trigger / persist the activation, and a virtual button in configurator is definitely an option.

jflyper, 2017-11-15:
@mikeller I was aware of the window, and yes, an MSP command would be a solution.

Here is a possible scenario.

- [U] Bind pulse must be entered manually or selected from a dropdown list. Receiver protocol may be used to narrow down the list elements.
- [U] "Bind" virtual button is pressed.
- [S] Configurator sends the MSP command, bind pulse count is saved.
- [S] User is instructed to power cycle the FC+receiver and perform bind operation on TX.
- [U] User power cycles the FC+receiver.
- [S] FC executes the bind pulse protocol.
- [U] User performs the bind operatin on TX.

Hmm... it saves one FC reboot compared to CLI method 😬

AndersHoglund, 2017-11-15:
All of above is true and not really different from current way of binding. Only difference, and also the main idea I think, is to be able to use a GUI drop down menu and button instead of the current cli sequence:

```
set spektrum_sat_bind = 9
save
```
All the rest is the same, power cycles, timing window , number of reboots (1) etc. 
You are making this issue more complicated then it really is. 

There could also be GUI option for the `spektrum_sat_bind_autoreset  = ON/OFF` option. And all the needed instructions on power cycle etc.

davecraige, 2017-11-16:
Yea, it might be good to simply have a button that says "Bind to Spektrum receiver with DSMX - and then a list of the different transmitters (like my DX6)

And possibly a second button that says "Bind to Spektrum receiver with DSM2" - and then have a list of all the transmitters that use that protocol.

This would help new users avoid the CLI.

jflyper, 2017-11-16:
I don't think it's worth the effort.

1. There are numerous how to video clips on how to do this.
2. There are boards that are not capable of sending bind pulses, so the button may not work.

martinbudden, 2017-11-17:
I agree, I don't think it is worth the effort. Especially since binding is an infrequent operation.

closed 2017-11-17 by martinbudden