betaflight/betaflight #4585: Omnibus F4 Nano LC SoftSerial problem

gsimpson2g, 2017-11-16:
I am in the process of trying to setup my Omnibus F4 Nano LC flight controller. It is running the OMNIBUSF4SD firmware. On 3.1.7, I can get Tramp Telemetry working with my Tramp HV using SoftSerial1 TX remapped to the LED_STRIP pin. After 3.2.0 and above, this is not working. Here is a copy of my DIFF. I am also looking to get SmartPort telemetry going on SoftSerial, but I have a feeling it wont work either right now.

'''
diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.2.2 Nov 11 2017 / 00:39:14 (77782e602) MSP API: 1.36

# name
name MADDRONER

# resources
resource MOTOR 1 A02
resource MOTOR 4 B00
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B06

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 0 1024 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1975 2050
aux 1 13 1 1525 2025
aux 2 28 0 1975 2050
aux 3 35 3 1975 2050

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_hardware = NONE
set min_check = 1020
set max_check = 1980
set rssi_channel = 7
set serialrx_provider = SBUS
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 270
set bat_capacity = 1200
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set current_meter = ESC
set battery_meter = ESC
set beeper_dshot_beacon_tone = 1
set pid_process_denom = 1
set osd_rssi_alarm = 45
set osd_cap_alarm = 1000
set osd_vbat_pos = 2423
set osd_rssi_pos = 2105
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2049
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2073
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_max_alt = ON
set osd_stat_endbatt = ON

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set anti_gravity_gain = 2500
set p_pitch = 49
set i_pitch = 55
set d_pitch = 33
set p_roll = 47
set i_roll = 55
set d_roll = 33
set p_yaw = 80
set i_yaw = 55
set d_yaw = 22

# rateprofile
rateprofile 0

set rc_rate = 55
set rc_rate_yaw = 70
set rc_expo = 2
set rc_expo_yaw = 2
set roll_srate = 74
set pitch_srate = 72
set yaw_srate = 72
set tpa_rate = 15

# 
'''

jflyper, 2017-11-16:
@gsimpson2g 
I've loaded your diff onto a Nano V3 test board, changed Tramp to SmartAudio (because I don't have Tramp), and it worked. So, the basic functionality of the softserial on PB6 (LED TH) is working.

Here's what I would do.

1. Full erase flash back to 3.1.7 and immediately configure only receiver, softserial and Tramp
This is to make sure the hardware is working.

2. Full erase flash to 3.2.2 and do the same.

FYI, this is an example configuration for Nano V3.
https://www.rcgroups.com/forums/showthread.php?2989357-OMNIBUS-F4-Nano-Wiring-Worksheet-and-Examples#post38643606

gsimpson2g, 2017-11-16:
Annnd it started working lol. I guess I needed to reboot the board again? Now to get SmartPort telemetry working. Thanks for looking at it jflyper, I was pulling my hair out last night trying to get it working.

closed 2017-11-16 by gsimpson2g

jflyper, 2017-11-16:
Happy props :simple_smile: