betaflight/betaflight #4586: Dshot commands buzzer

Bugg69s, 2017-11-16:
Just a question. 
Did anyone fixed the buzzer on omnibus ?

wind0r, 2017-11-16:
what is there to fix?

mikeller, 2017-11-17:
Is that the issue with the OMNIBUS permanently beeping when rebooting and flashing? If so, that's a design limitation, and there is no software fix.

Bugg69s, 2017-11-17:
@Mikeller no the dshot buzzer, i don't think it works on omnibus boards. On Betaflight 3.2 RC5 it worked. But after updating to 3.2, 3.2.1 or 3.2.2 it do not work at all.

jflyper, 2017-11-17:
@Bugg69s It works. Probably requires up to date blheli firmware.

wind0r, 2017-11-17:
@jflyper or just enabling it ;) e.g set  beeper_dshot_beacon_tone to another value then 0

jflyper, 2017-11-17:
Yeah. That's basic. I though default has 5 or something?

wind0r, 2017-11-17:
@jflyper  No. default ist 0 (off)

@Bugg69s  solong you can run dshot everything is working, dshot cmd can not workingfor just one target. that is not how software works. this is just a settings failure.

jflyper, 2017-11-17:
Ah, it was zero.

Bugg69s, 2017-11-17:
Ok, thanks :)

closed 2017-11-17 by Bugg69s