betaflight/betaflight #4596: 3D Mode out of control arming issue w/logs

DungAdumbs, 2017-11-17:
Having random out of control arming issue w/ 3D mode.  Seems to requite power cycle to clear.  Has not happened after a successful arm.

Have documented to the best of my ability with logs, videos & cli dump at:  https://www.rcgroups.com/forums/showpost.php?p=38648352&postcount=4

DungAdumbs, 2017-11-19:
Had an incident on rearm after a successful arm.

Reviewing logs, motor 3 arms at higher throttle consistently when problem occurs.

Are there Devs interested in improving 3D mode?  Seems requests fall on deaf ears.... The above issue is really a safety issue.

DanNixon, 2017-11-19:
> Seems requests fall on deaf ears....

No. Requests fall on busy ears.
Betaflight is a hobby project for almost all developers. It is not a full time job and doesn't pay a wage.
2 days is not enough time to assume no one is interested the issue you reported.

I can't see anything obvious that would be causing the issues you see on arming, this is going to require further investigation.
As for 3D transitions, there have already been ramped transitions similar to what you propose in your RCG post proposed: #4354.

DungAdumbs, 2017-11-19:
Sorry, very poor choice of words on my part.  New to Github, first request was handled super fast... this is my second.  I am extremely grateful to all those behind this project all the way back to multiwii.  The tireless efforts of a few have benefited so many.  Thank you and all that have come before.

I've dabbled in 3D mode for years with poor success.  Current builds are much better but still need some finish work. I've watched 3D issues just fade away on the main RCG thread.  I see 3D mode of benefit to most if could become seamless and reliable even if users only use a fraction of it's power.  I'm super excited to see how close we are now and want to help polish things up.  I'm very happy to see that 3D does matter!!

Thank you!!

DanNixon, 2017-11-19:
No worries.

The problem is very few (if any) active developers fly 3D so problems usually go unnoticed for a long time. Of course getting feedback is an important part of addressing the remaining issues.

fftunes, 2017-12-05:
> Reviewing logs, motor 3 arms at higher throttle consistently when problem occurs.

Could this have anything to do with dshot timer issues mentioned elsewhere?

DungAdumbs, 2017-12-05:
I'm on the road... I'm betting you are seeing my bias for right roll and down elevator.

mikeller, 2017-12-05:
@fftunes: Unlikely. Dshot uses a checksum, so if there was a timer / DMA problem, the resulting frame would be discarded by the ESC because of an invalid checksum.