betaflight/betaflight #4597: unit test do not build using clang++ 3.4

s-h-x, 2017-11-17:
Some headers are defining structs containing const members, and some compilers (in my case : Ubuntu clang version 3.4-1ubuntu3  ) will not support initialisation of such code (which happens when building tests)

compiling unit/cli_unittest.cc 
unit/cli_unittest.cc:93:29: error: call to implicitly-deleted default constructor of 'const cliValueConfig_t'
    const clivalue_t cval = {
                            ^
../main/interface/settings.h:125:28: note: default constructor of '' is implicitly deleted because field 'lookup' has a deleted default constructor
    cliLookupTableConfig_t lookup;
                           ^

Instead, defining those member as being "const" should be conditionned by the presence of cclang feature cxx_aggregate_nsdmi :
https://clang.llvm.org/docs/LanguageExtensions.html#c-14-default-initializers-for-aggregates

DieHertz, 2017-11-17:
You seem to be compiling C code as C++. C doesn't have the notion of constructor altogether

s-h-x, 2017-11-17:
unit tests are c++, compiled with clang++, an din this context, some compilers (versions) do not allow to initialize statically const members (like the init of clivalue_t in cli_unittests.cc for instance)
Partial (after a revert of my patch) log of a failed  build of tests :
[testbuild.log](https://github.com/betaflight/betaflight/files/1482793/testbuild.log)
Thanks for your feedback
Steven

wind0r, 2017-11-17:
The Buildservers (tarvisci and jenkins) are using clang without a Problem. You may should have a look there.

AndersHoglund, 2017-11-17:
Please make your env a bit more up-to-date date. This works:
$ clang++ --version
clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

s-h-x, 2017-11-17:
I did it, and the version in use is 3.9, and I'm this version the
implements of aggregate unit had changed.
My changeset is meant to support older compiler (mine is Ubuntu 14.04 Lts
default)

s-h-x, 2017-11-17:
I wish I could, but I can't administrate this machine. But yes, I could
rebuild the compiler from sources ....

s-h-x, 2017-11-17:
(Sorry for the numerous typos, I'm writing from my phone)