betaflight/betaflight #4601: spektrum satellite will not go into bind mode using known CLI's

sherer46, 2017-11-17:
Hi, 
Problem: DSMX Satellite blinks once upon power up, will not go into bind mode using the CLI's. I have tried 3 different satellites. 
I just got the Furibee x215.
It has the Kakute f4 board, not sure if it's a clone or not, the actual board looks different from the board in the user manual.
Satellite is soldered with the Orange wire onto 3.3v, black to Ground, and grey to "r3" correctly with corresponding wires- tested with a multimeter, the satellite does receive power  (3.3v).
Under Ports, Uart 3 set up to serial rx. I have the configurations set up to speksat and 2048.
I have tried every way possible that I have come across, Using the set spektrum_sat_bind=9 and trying the set spektrum_sat_bind_autoreset= ON and OFF, no difference.
I have reset drivers several times.
I cannot get the Satellite to go into bind mode no matter what I try. I am brand new to Betaflight and have read something about Uart reversal or something and don't really understand it.

# version
# Betaflight / KAKUTEF4 (KTV1) 3.2.2 Nov 11 2017 / 00:32:18 (77782e602) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ARMED
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set serialrx_provider = SPEK2048
set spektrum_sat_bind_autoreset = OFF
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 7
set failsafe_procedure = AUTO-LAND
set pid_process_denom = 1

# profile
profile 0

# rateprofile
rateprofile 0

mikeller, 2017-11-18:
Hi @sherer46. Looking at the specification for the KAKUTEF4, this board does have a switchable inverter on UART3. These inverters can sometimes interfere with Spektrum binding. Can you try using UART6 or UART1 for the serial RX instead? (You'll have to change that in the 'Ports' tab in configurator.)

jflyper, 2017-11-18:
Yes, the uni-directional inverter prevents bind pulses to propagate from FC to receiver.

There may be a trick to use RX3, but please try other (naked) UARTs to verify the integrity of hardware and correct procedure.

de-apu, 2018-05-10:
I have the same problem! My Spektrum SPM9645 Remote DSMX Receiver does not go into bind mode with my MATEKSYS F405-OSD. I tried all URAT Ports and all combinationes with spectrum_sat_bind settings, but no effect.
Any suggestions?

closed 2018-06-16 by stale[bot]