betaflight/betaflight #4602: dshot commands exit turtle mode.

Jahnkeanater, 2017-11-18:
If I enable turtle mode and then switch to esc beeper mode while the quad is armed disarming and then rearming the quad while the beeper is enabled does not undo reverse motors from turtle mode before arming the quad. This causes the quad to yaw spin violently.

In this case, it seems like the FC never sends the dshot command to return the motors to the normal direction.

1) enable turtle mode.
2) arm.
3) switch from turtle mode and enable esc beeper.  "I have this on a 3 way switch"
4) hide behind your blast shield.
5) toggle the arm switch off then on.
6) quickly disarm.

I'm pretty sure this quad is running BF 3.2.0.

fftunes, 2017-11-19:
Imho it shouldn't be allowed to enable esc beeper while being armed, and the other way around, to arm while esc beeper is on.

mikeller, 2017-11-19:
@fftunes: If ESC beeper is enabled while armed, the beeper commands are not sent until the flight controller is disarmed. I suspect this is caused by the ESC beeper commands commencing too soon after disarming, thus causing the ESCs to not properly receive the 'reverse direction' command.

Jahnkeanater, 2017-11-20:
Whats the best way to make sure the turtle mode commands are sent before the beeper commands?

Would a simple delay before executing the beeper command work well enough?

You can't simply disable the beeper because someone might want to enable the hardware beeper in flight. Disabling arming while the beeper is enabled wouldn't prevent this since the issue is disarming and enabling the beeper simultaneously. Even if you turn the beeper off the motors will still spin the wrong direction.

dthubereng, 2017-12-08:
When I was flying this past weekend I noticed some strange behavior with "flip over after crash" mode. I currently have the mode enabled using the same Aux channel as the enable for my ESC beeper mode. The expectation was that when I crash I can disarm and toggle into "flip over" mode to sound the ESC beeper or re-arm to flip it back over, "knowing" that the beeper would stop when armed. I can follow up with this later to see if I can find something that causes an issue between "flip over" and "beeper" modes (my situation sounds similar to what @Jahnkeanater described in his OP).