betaflight/betaflight #4608: Multiple OSD profile/layout feature request

1337flite, 2017-11-18:
This is a feature request to have multiple OSD layouts similar to the way you can have different PID/rate profiles.

Use case is when flying normally I want a minimal OSD config, but when PID tuning, trouble shooting etc  I want more info on the screen and again I'd want a different layout when flying a quad with a barometer, magnometer or GPS fitted.

Ideally it should be possible to select which profile/layout is active via a switch.

Thanks - you BF devs rock.

etracer65, 2017-11-18:
This sounds like a good idea!  Possibly make the OSD settings/layout profile dependent.  You could then switch profiles using stick commands while disarmed.  Also in the mentioned use-case (tuning) it would make sense to tune in a different profile in isolation from your main working configuration.

DanNixon, 2017-11-18:
Duplicate of #4155

closed 2017-11-18 by mikeller