betaflight/betaflight #4609: WindUp and oscillations by latest changes in ITerm calculations

AirBreak69, 2017-11-18:
I used release 3.2.2 on a micro quad with AIRBOTF4 firmware: OK.
Changed to build 1820 which contained #4587 dT changes: Wind up before lift off.
Changed to build 1822 which contained #4594 ITerm changes: Rising roll oscillations on lift off within some seconds.
Sorry, i can't provide any BB log. The FC doesn't have external flash.
I don't want to try that code on bigger quads...
@martinbudden @mikeller Please review or revert this particular PRs

martinbudden, 2017-11-18:
@AirBreak69 , thanks for your report. I'm actively working on this change.

Can you try build 1822 (or later) again, but increase your D values. Try going all the way up to the max (ie 200) and report back on your findings. Thanks.

AirBreak69, 2017-11-18:
Using Build 1822, i started with roll/pitch D's of around 33: Oscillation after approx. 3s
Increased D's to 66: Oscillation directly after lift off.
Increased D's to 100: Faster oscillation directly after lift off.
Increased D's to 150: Oscillation gets wilder.
Increased D's to 200: same as 150.
The controlability didn't change much on the higher D's. For safety I stopped when oscillation passed ±20° after some seconds - mainly on roll axis.
@martinbudden I hope that this helps your thoughts.

HFMan, 2017-11-18:
@AirBreak69 - I reported the same issue in #4594 - The problem is that the PID_D values are always zero- you never have any D correction no matter how high you have it set.  My Blackbox log shows this.

AirBreak69, 2017-11-19:
Can confirm that #4616 (build 1827 tested) fixes this issue.
Thanks.

closed 2017-11-19 by AirBreak69