betaflight/betaflight #4613: Solution for "calib" flag when board wont arm whilst connected to configurator. BF3.2.2

chaotixtech, 2017-11-18:
I've just advised another person to set moron_threshold to 90 to cure the calib flag reason for not arming after flashing 3.2.2. 
Is this reported as an issue yet as it's cropped up a few times now. 
Previously working boards reporting arming flag calib when connected to configurator. I first had it with an Alienwhoop F4 and the most recent was on a Luminier F4 aio.

mikeller, 2017-11-19:
The level of noise that `moron_threshold` is checked against is dependent on the model of gyro, and to some extent on the condition of the gyro. The current default values seem to work for most users. Sometimes a crash will damage a gyro to a point where it still works, but the noise level is increased, thus making it necessary to increase the value of `moron_threshold`.

chaotixtech, 2017-11-19:
I get how a change in the gyro, physically, could raise the relative noise floor but in most cases the boards were behaving fine on earlier firmware and only required the change after a flash. I've just now had a report of a brand new out the box clracing F4 needing the setting raised and functioning correctly afterwards. In my case I chose 90 as a value as it was roughly double the default and have since been recommending that with success. I wonder if is just the fact of changing the value that fixes it, perhaps some targets default to zero after a flash? 
I'm trying to gather as much info as I can, is there anything I should add?

dassdrones, 2017-11-26:
This worked on my xracer F4 board, although I had to set moron_threshold to 120. Also seemed to correct an issue with having to rearm multiple times in flip-over after crash mode to get the motors to reverse correctly.

chaotixtech, 2017-11-27:
I think a higher default needs to be set for moron_threshold. Over the weekend I've come across another dozen or so cases of raising it being the solution to remove the cailb flag and allowing arming. Around half have been on new F4 boards and the rest on boards that were working before an upgrade flash. I think it's getting regular enough that the default needs changed or the code that's started triggering it needs looked at. 
Word is going about on raising moron_threshold as a cure, along with many raised eyebrows at the naming!

mikeller, 2017-11-27:
Can somebody verify that this is indeed the firmware update by flashing the old version back and checking if the lower `moron_threshold` is still working? An increase in noise level is often caused by damage / degradation to the gyro caused by hard impacts.

brucesdad13, 2017-11-27:
For what it's worth, I've pulled off and replaced at least 30 brand new gyros both MPU-6500 and MPU-9250 on handmade AlienWhoops assuming I received a bad batch for gyros or perhaps that my soldering wasn't perfect. @jflyper and I though even perhaps it was related to SPI clock. They always work fine at 8/8 but once 32kHz enabled won't clear calibration flag. I've named them "sad gyro" fcs and sent to friends. It happens on F7 also... not just F4. And doesn't matter overclocked or not... I also tried changing SPI clock divisor in the code. The datasheet says 1MHz to 20MHz and F4 is running around 21MHz. Sometimes after replacing 3 or 4 in succession I would get one to work at 32/32. I've just been plodding along... will test moron threshhold on some of my "sad" fcs and see if that helps.
[image]

brucesdad13, 2017-11-27:
Setting moron threshold to 200 I am able to get one of the sad fcs to run at 32/32

mikeller, 2017-11-27:
@brucesdad13: One thing to say there is that the 32 kHz mode is not officially a supported operation mode for these gyros. So if they work at 32 kHz - good. If they don't (i.e. data is too noisy to be used), that does not put the device outside of the manufacturer's spec for a working device.
I've had at least one case of a MPU-9250 that worked fine on 32 kHz when the flight controller was new. After a crash, the noise level at 32 kHz was increased to a point where it wasn't possible any more to get the gyro to calibrate. After changing back to 8 kHz, everything is fine again, and I am still flying this board.

brucesdad13, 2017-11-27:
@mikeller ah that's true. 😄

stepansonsky, 2018-05-06:
It happens to me as well on BF F4, when I want to arm, calib tag shows on OSD, sometimes it works after several tries, sometimes after restart. Ideas?

stepansonsky, 2018-05-27:
Seems to be fixed with moron_threshold = 200. Can capacitor fix this on default value?

mmakam2, 2018-06-13:
im having this issue on a RevoltOSD thats brand new, ive lowered it to 16khz/16khz and raised calib noise limit to 200, still getting the arming code 12 issue.

mikeller, 2018-06-13:
@mmakam2: In some cases the high noise level is caused by not enough filtering on the supply voltage for the MPU. Additional capacitors can help, check RCGroups for hints.

mmakam2, 2018-06-13:
I have added a Panasonic 470uf low esr cap to the xt60, still have the same issue.

It arms fine in 8khz mode.

closed 2018-07-20 by stale[bot]