betaflight/betaflight #4622: New 3.2.2 GUI shows Filtering turn on?

claytonbakerjr, 2017-11-19:
Hi
Please forgive if this was brought up under a different title and if so please point me to it and close this one.
I have 5 quads all flashed with BF 3.2.0 (I have not gone to latest yet) 
One of those quads was just recently flashed with 3.2.0 less than a week ago
yet the others have been flashed since the Official Release.
In the NEW GUI 3.2.2 in the Filter Settings screen all Sliders are enabled, Which from my understanding of this new slider thing, is that now none of my filters are disabled (set to zero). As they were before in the previous GUI version. 
In my previous experience (unless I did not pay attention) I do not recall a GUI version change,  that made changes to the firmware settings from a previous flashed board, it typically read the flashed setup and moved forward??
To me I feel the GUI should have read those settings and nothing would have changed.
Could someone please enlighten me on why I see this? BTW I have others that have experienced this too.
To add, in all cases the Slider for Dynamic Filtering was still enabled (as before) as well as PT1 in the Filtering Settings screen, the only change was all filters were turned back on (Slider enabled)
In my case I have the following boards flashed with 3.2.0
CLRACINGF4 (3 quads)
COLIBRI_RACE (1 quad, recently flashed)
FF_PIKOBLX (1 quad)

claytonbakerjr, 2017-11-19:
To add, I did a CLI on 2 of the 5 quads and it shows that all of my notch filters are at zero. So it seems the GUI is not picking that up unless we have a misunderstanding on the Sliders??

claytonbakerjr, 2017-11-19:
Doing some more research I have found that after doing a diff of original settings and then moving sliders and then another diff, comparing the 2 I now see 2 commands that do not exist in the original diff file.
set gyro_notch1_cutoff = 0
set gyro_notch2_cutoff = 0
I would (guess?) this might be the reason the gui is not showing
what the CLI does?

etracer65, 2017-11-19:
This is just configurator wonkiness.  The classic CLI method of disabling the notch only required that the center frequency be set to zero - the cutoff didn't matter.  If the center frequency was zero then the notch was disabled.

The problem is that the configurator only considers the notch disabled if *both* the center and cutoff are zero so it falsely shows the notch enabled.  To further complicate issues, the configurator replaces the zero value with the default since it presumes the filter is enabled and you therefore can't have a zero value for the center.  If you then click the "save" button on the filters tab then the notch will be re-enabled with a default value for the center and with whatever you had in the cutoff previously.  It looks very confusing!

You can minimize the confusion by switching to setting both the center and cutoff to zero to disable the notch.  Then the configurator will show the slider as "off".  However if you switch the slider back to "on" then the configurator will populate both the center and cutoff with default values (nothing else it can really do).

mikeller, 2017-11-19:
@basdelfos: Can you please look into this?

basdelfos, 2017-11-20:
Will look into this tonight.
Solution is to let the configurator not look at both center and cutoff to disable the checkbox, but only when center is 0 and ignore the cutoff. When cutoff is not 0 and the checkbox is turned on, the cutoff value must not be reset to default.

I still would find it much more user friendly to enable the notch filters with there own setting and leave the center and cuttoff values alone, like the initial plan I had. Then all this confusion and wonkiness would not be present.

mikeller, 2017-11-20:
> I still would find it much more user friendly to enable the notch filters with there own setting and leave the center and cuttoff values alone, like the initial plan I had. Then all this confusion and wonkiness would not be present.

Nobody said that this should not be done - just that it should be done in the configurator, and not in the firmware. If we add on/off switches to every setting in the firmware that currently works as '0 means off', the size of the configuration will increase dramatically.

claytonbakerjr, 2017-11-20:
Thanks for looking into this everyone

closed 2017-11-21 by mikeller