betaflight/betaflight #4623: Matek F405 MINI 32MB dataflash records duplicate logs and fills up quickly.

kicekilla, 2017-11-19:
Recently finished my build with the new Matek F405 mini that has Winbond W25Q256 flash chip (32MB). My problem I am having is that every flight is creating duplicate logs so the dataflsh fills up quickly. A simple hover test at even 500Hz fills about have of the flash. I have not heard back from Matek or heard any other solution from other sources (rcgroups, facebook). 

Matek F405 mini
ori32 blheli32 running dshot1200
Betaflight 3.2.2
same issue with 16/16, 16/8, 8/8, 8/4
bb set to spiflash at 500hz, also happens with 1 and 1.5 khz... 

F405 mini is powered from Matek vtx hv BEC (acceptable per Matek website). 

Not sure how it can be duplicated... i have tried reflashing a few times, but cannot try another ver. of firmware as the 32MB flash is only supported in 3.2.2

'''
# diff all

# version
# Betaflight / MATEKF405 (MKF4) 3.2.2 Nov 11 2017 / 00:35:15 (77782e602) MSP API: 1.36

# reset configuration to default settings
defaults nosave

# name
name HOT_START

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature MOTOR_STOP
feature LED_STRIP
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial
serial 0 1024 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 32 115200 57600 0 115200

# led
led 0 0,0::A:9
led 1 0,1::A:0
led 2 0,2::A:0
led 3 0,3::A:0
led 4 1,3::C:10
led 5 2,3::C:10
led 6 3,3::C:10
led 7 4,3::CB:10
led 8 5,3::CB:10
led 9 6,3::CB:10
led 10 7,3::CB:10
led 11 8,3::CB:10
led 12 9,3::CB:10
led 13 10,3::CB:10
led 14 11,3::CB:10
led 15 12,3::C:10
led 16 13,3::C:10
led 17 14,3::C:10
led 18 15,3::A:9
led 19 15,4::A:9
led 20 15,5::A:9
led 21 15,6::A:0

# color

# mode_color
mode_color 6 0 2

# aux
aux 0 0 1 1900 2100
aux 1 15 0 900 1700
aux 2 28 0 1300 2100
aux 3 35 4 900 1100
aux 4 36 2 900 1100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 260
set gyro_notch2_cutoff = 150
set gyro_use_32khz = ON
set baro_hardware = NONE
set rssi_channel = 8
set sbus_inversion = OFF
set blackbox_p_ratio = 48
set blackbox_device = SPIFLASH
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT1200
set current_meter = ESC
set ibatv_scale = 300
set yaw_motors_reversed = ON
set deadband = 4
set yaw_deadband = 6
set pid_process_denom = 2
set tlm_switch = ON
set pid_in_tlm = ON
set osd_tim2 = 1281
set osd_vbat_pos = 12
set osd_rssi_pos = 2075
set osd_tim_1_pos = 504
set osd_tim_2_pos = 2455
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2424
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2048
set osd_mah_drawn_pos = 2080
set osd_craft_name_pos = 2432
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2060
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set debug_mode = NOTCH
set vcd_video_system = 2
set esc_sensor_halfduplex = ON

# profile
profile 0

set dterm_lowpass_type = PT1
set vbat_pid_gain = ON
set anti_gravity_gain = 2500
set setpoint_relax_ratio = 60
set dterm_setpoint_weight = 80
set crash_recovery = ON
set p_pitch = 57
set i_pitch = 41
set d_pitch = 11
set p_roll = 42
set i_roll = 33
set d_roll = 9
set p_yaw = 99
set i_yaw = 50

# profile
profile 1

set dterm_lowpass_type = PT1
set vbat_pid_gain = ON
set anti_gravity_gain = 3500
set setpoint_relax_ratio = 60
set dterm_setpoint_weight = 80
set crash_gthreshold = 600
set crash_recovery = ON
set p_pitch = 57
set i_pitch = 41
set d_pitch = 10
set p_roll = 42
set i_roll = 33
set d_roll = 9
set i_yaw = 50
set d_yaw = 15

# profile
profile 2

set dterm_lowpass_type = PT1

# restore original profile selection
profile 1

# rateprofile
rateprofile 0

set rc_rate_yaw = 102
set rc_expo = 30
set rc_expo_yaw = 30
set tpa_rate = 13
set tpa_breakpoint = 1550

# rateprofile
rateprofile 1

set rc_rate_yaw = 102
set rc_expo = 30
set rc_expo_yaw = 30

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
'''

jflyper, 2017-11-19:
I'll look into this.

jflyper, 2017-11-21:
@kicekilla 

Can't reproduce the problem.

Here, I run 1KHz logging for 300 secs, coming out with 1.5MB log file without any duplicate content.

kicekilla, 2017-11-21:
Hmmm... yeah I don't know what is up with it.. I have tried reflashing... maybe ill try to reflash to a 3.3 nightly. 

Any other suggestions?

jflyper, 2017-11-21:
Oh, wait... There's something wrong with 3.2.2...

jflyper, 2017-11-21:
I realized that I was doing my test with something newer (says 3.3).
As soon as I flashed with 3.2.2, remaining capacity is already 14.3MB.
Let me look into this further.

kicekilla, 2017-11-21:
Tomorrow I’ll flash to the latest nightly and see if there is a difference.

jflyper, 2017-11-21:
There's something definitely wrong, including 3.3 (current master).

It's writing to somewhere in a middle of the unused region, causing "detect last page written" algorithm to fail.

kicekilla, 2017-11-21:
Roger, I’ll standby.

jflyper, 2017-11-21:
It will take a while. Just a hunch, but I think there are some incompatibilities between the base M25P16 chip and the W25Q256 chip.

jflyper, 2017-11-21:
Hehe, flash chip larger than 128Mb (16MB) is not compatible with code written for smaller chips, as address exceeds 3 bytes beyond 128Mb (16MB).

jflyper, 2017-11-21:
@kicekilla Can you try this?

[betaflight_3.3.0_MATEKF405-4byte_address.hex.zip](https://github.com/betaflight/betaflight/files/1491397/betaflight_3.3.0_MATEKF405-4byte_address.hex.zip)

kicekilla, 2017-11-21:
Yeah I’ll they it today!

kicekilla, 2017-11-21:
seems to have fixed the issue of filling up quickly and duplicate logs per flight... : ) so Thats awesome! there still seems to be some issues. I have tried logging at .5, 1, 1.5, 2KHz and spektral analyzer is showing a little weird and 1.5khz seems to have an error. see the attached pictures with spektral analyzer open on the different logging rates(note the sampling rate at the bottom for 1.5khz. data is sporadic. 

.5K
[image]
1K
[image]
1.5K
[image]
[image]
2K
[image]

jflyper, 2017-11-22:
@kicekilla How about this one?

[betaflight_3.3.0_MATEKF405-4byte_address2.hex.zip](https://github.com/betaflight/betaflight/files/1496865/betaflight_3.3.0_MATEKF405-4byte_address2.hex.zip)

kicekilla, 2017-11-22:
@jflyper ill give it a try... might be a few days till i get back in town.. also realized last night that the OSD page on configurator is blank on the prevous hex you posted... could just be me not using the most recent configurator for nightly builds... im on 3.2.2 configurator.

jflyper, 2017-11-22:
May be the configurator problem. Pull one from github. Mine says 10.0.0.

PR #4635 (2017-11-27): FLASH: Use 4B address for device > 128Mb (16MB)
PR status: Ready to merge

Fixes: #4623

Large NOR flash devices, namely those over 128Mb capacity require extended (more than 3 bytes) addressing.

This PR implements "Enter 4B address mode" command (0xB7), which is common among
- Winbond W25Q256FV, W25Q256JV and W25Q512JV
- Macronix MX25L25635E
and use 4-byte addressing if the size of the device is larger than 128Mb.
closed 2017-11-27 by mikeller