betaflight/betaflight #4626: DShot1200 Frame corruption occurring mid-throttle

TonyBazz, 2017-11-20:
Ive been noticing weird flight behaviour, including non-linearity and spasms, occurring mid-flight on Dshot1200. Dshot600 and Proshot1000 work perfectly fine. Its happening on BLUEJAYF4, FORTINIF4 and SPRACING F3 (All my FC's), havent tested other FCs but im assuming its a common issue.

So i scoped the output of the FC and seen 17 pulses on the output at certain points in the throttle range.

Video link showing corrupted frame: https://youtu.be/z6AdLGMXc7M

TonyBazz, 2017-11-20:
Further to my post,..
- Issue is present on BF3.2 & BF3.3. 
- Does not occur on all motor outputs. Seems random.

knockoutperformance, 2017-11-20:
I had to remap timers and pins on an spracingf3 to run dshot 1200, there are timers conflicts. I have that target file correction on github under my account. SPRACINGF3_DSHOT1200. I would think the other two you have are also having conflicts.

TonyBazz, 2017-11-20:
Why would dshot600 and proshot work then? Uses the same timers just clocked differently.

knockoutperformance, 2017-11-20:
I'm not technical enough to answer that. I noticed timing a issue when on dshot 600(could actually here it in one motor), that's why I remapped(pulled data sheet and looked at DMA timers), ended up being able to use dshot1200 afterwards. I believe UART timer conflicts, I would have to look, I use a dysf4pro now. I had to move motor 3&4 to motor 5&6 and change the DMA timers for those.

mikeller, 2017-11-20:
It _could_ be timer / DMA conflicts. I've seen similar things happen to DShot outputs when they were using DMA channels that were used by SPI as well.

conkerkh, 2017-11-23:
According to my knowledge DMA was never used on SPI in BF. Could be that timer has wrong PSC and ARR register set up for DShot1200 some timers are 16Bit others 32Bit so it’s not that obvious. It would be good if you could hook up logic analyser, and have a look on bench what’s going (without props lol)

TonyBazz, 2017-11-23:
@conkerkh I did hook it up on the bench. Check out the video link in the first post.

conkerkh, 2017-11-23:
Sorry didn’t notice, it’s interesting I think it’s code related, will have a look, but static analysis should catch up something like that

jflyper, 2017-11-23:
Just a hunch, but it sounds like timer output being disabled prematurely upon processing DMA completion. It happens intermittently because of the DMA completion interrupt timing.
Thoughts?

TonyBazz, 2017-11-23:
Ive been tinkering with the code to...
what ive found so far.. When i limit the Dshot frame to 15 bits, removing the LSB, the frame seems good up to 3.6Mhz.

TonyBazz, 2017-11-23:
@jflyper Thats a very plausible theory! Maybe the DMA completion interrupt is taking too long to execute and needs optimising... or probably just a higher priority.

ledvinap, 2017-11-23:
@jflyper : IMO 16 pulses are expected, 17th pulse is error. So disabling too early is probably not the cause.

@TonyBazz : The system should be deterministic. Increasing priority will only mask the problem.
Writing zero as last value should disable pulse generation indefinitely, so there should be enough time to handle DMA.

TonyBazz, 2017-11-23:
@ledvinap 

I increased dshotbuffer size to 20 bytes with last 4 bytes as zero. Same thing no change. 
Really weird issue. :/

knockoutperformance, 2017-11-23:
i looked at the old files i used to get dshot1200 working on an spracingf3.

 can i just post the target.c and target.h here?

 i had to change timers around due to conflicts, tried many different combos of pins and timers until i ended up with what i had. pb10 and pb11 pins use uart3 timers.

i was asked before why to run dshot1200 on spracingf3 when i opened issue, so it was closed awhile back.

the issue is also apperant/audible in dshot600 on the spracingf3 without remapping.

TonyBazz, 2017-11-23:
@knockoutperformance 
Feel free to post them here.. but i still think we need to really understand what is happening. 
The issue is occurring on multiple boards.. We need to find a global solution.

knockoutperformance, 2017-11-23:
ok, i understand it is happening on a couple boards. fixed dshot1200 for an x_racerspi f3 board also(almost the same as sprf3, but with spi gyro).

altered files spracinf3:
   target.c
`//modded By Kev o. to accomadte the sprf3 board dshot1200
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
// ppm
    DEF_TIM(TIM2, CH1, PA0, TIM_USE_PPM, 0),

// Motors 1-4
    DEF_TIM(TIM16,CH1, PA6, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED),  //mtr1  -sprf3 m1 pin                       DMA(1,6)
    DEF_TIM(TIM8, CH1N,PA7, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED | TIMER_OUTPUT_INVERTED), //mtr2 -sprf3 m2 pin DMA(2,3)
    DEF_TIM(TIM8, CH2, PB8, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED),  //mtr3  -sprf3 m5 pin                       DMA(2,5)
    DEF_TIM(TIM17,CH1, PB9, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED),  //mtr4  -sprf3 m6 pin                       DMA(1,7)

// Motors 5-6 or SoftSerial
    DEF_TIM(TIM3, CH3, PB0, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED),  //mtr5  -sprf3 SONAR_TRIGGER_PIN            DMA(1,2)
    DEF_TIM(TIM3, CH4, PB1, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED),  //mtr6  -sprf3 SONAR_ECHO_PIN               DMA(1,3)
 
// Motors 7-8 or UART2
    DEF_TIM(TIM2, CH4, PA3, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED),  //mtr7  -sprf3 m7 pin   UART2_RX
    DEF_TIM(TIM2, CH3, PA2, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED),  //mtr8  -sprf3 m8 pin   UART2_TX
 
// No LED for Hexa-Dshot; DMA conflict Motor 5 (PB0); PPM if not used.
    DEF_TIM(TIM1, CH1, PA8, TIM_USE_LED, TIMER_OUTPUT_ENABLED),  //LED_STRIP  	DMA(1,2)
	
};`

target.h
` //modded By Kev o. to accomadte the sprf3 board dshot1200

#pragma once

#define TARGET_BOARD_IDENTIFIER "SRF3"

#define CONFIG_FASTLOOP_PREFERRED_ACC ACC_NONE
#define LED0_PIN                PB3

#define BEEPER                  PC15
#define BEEPER_INVERTED

#define USE_EXTI
#define MPU_INT_EXTI            PC13
#define USE_MPU_DATA_READY_SIGNAL
#define ENSURE_MPU_DATA_READY_IS_LOW

#define GYRO
#define USE_GYRO_MPU6050
#define GYRO_MPU6050_ALIGN      CW270_DEG

#define ACC
#define USE_ACC_MPU6050
#define ACC_MPU6050_ALIGN       CW270_DEG

#define BARO
#define USE_BARO_MS5611
#define SONAR
#define SONAR_TRIGGER_PIN       PB0
#define SONAR_ECHO_PIN          PB1
#define SONAR_SOFTSERIAL1_EXCLUSIVE
#define MAG
#define USE_MAG_HMC5883
#define MAG_HMC5883_ALIGN       CW270_DEG
#define USE_MAG_DATA_READY_SIGNAL
#define ENSURE_MAG_DATA_READY_IS_HIGH
#define MAG_INT_EXTI            PC14

#define USE_UART1
#define USE_UART2
#define USE_UART3

#define USE_SOFTSERIAL1
#define SOFTSERIAL1_RX_PIN      PB0  // PWM 5
#define SOFTSERIAL1_TX_PIN      PB1  // PWM 6
//#define USE_SOFTSERIAL2
#define SERIAL_PORT_COUNT       4    // 5 if softserial2

#define UART1_TX_PIN            PA9
#define UART1_RX_PIN            PA10

#define UART2_TX_PIN            PA14 // PA14 / SWCLK
#define UART2_RX_PIN            PA15

#define UART3_TX_PIN            PB10 // PB10
#define UART3_RX_PIN            PB11 // PB11

#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C_DEVICE              (I2CDEV_1)

#define USE_SPI
#define USE_SPI_DEVICE_2 // PB12,13,14,15 on AF5

#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_CS_PIN           PB12
#define M25P16_SPI_INSTANCE     SPI2
#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT

#define DEFAULT_VOLTAGE_METER_SOURCE VOLTAGE_METER_ESC  // wraith32 35a esc telemetry
#define DEFAULT_CURRENT_METER_SOURCE CURRENT_METER_ESC  // wraith32 35a esc telemetry

#define USE_ADC
#define ADC_INSTANCE            ADC2
#define VBAT_ADC_PIN            PA4
#define USE_MSP_CURRENT_METER
#define CURRENT_METER_ADC_PIN   PA5
#define RSSI_ADC_PIN            PB2

#define OSD
#define USE_OSD_OVER_MSP_DISPLAYPORT
#define USE_SLOW_MSP_DISPLAYPORT_RATE_WHEN_UNARMED

#define USE_ESCSERIAL
#define ESCSERIAL_TIMER_TX_PIN  PA0 // unsure what this even does
#define USE_ESC_SENSOR

#define REMAP_TIM16_DMA
#define REMAP_TIM17_DMA

#define USE_SERIAL_4WAY_BLHELI_INTERFACE

// IO - stm32f303cc in 48pin package
#define TARGET_IO_PORTA         0xffff
#define TARGET_IO_PORTB         0xffff
#define TARGET_IO_PORTC         (BIT(13)|BIT(14)|BIT(15))
#define TARGET_IO_PORTF         (BIT(0)|BIT(1)|BIT(3)|BIT(4))

#define USABLE_TIMER_CHANNEL_COUNT 10
#define USED_TIMERS             ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(8) | TIM_N(15) | TIM_N(16) | TIM_N(17) )`

i probably did not post this right. 
but these changes got me running nice and smooth dshot1200, with idle down to 1014.

ledvinap, 2017-11-23:
@TonyBazz : Maybe try delaying the interrupt artificially - it may make the problem more predictable ...

conkerkh, 2017-11-23:
How about creating timer irq, counting how many times it overflows and disabling DMA transfer after alll bits are sent? I would love to try something but no accesss to board with debug port atm...

ledvinap, 2017-11-23:
@conkerkh : DMA does stop automatically. It seems that there is some bug hidden somewhere. 

Maybe DMA is starved of bus access and updates registers too late  - this may be visible even on other bits, probably on long -> short pulse sequence

jflyper, 2017-11-24:
@ledvinap You mean underrun?

jflyper, 2017-11-24:
@here 
FYI, there is another case of short frames which is not resolved yet.
F4 Dshot with N channels on advanced timers (TIM1 & TIM8) don't work #3549

conkerkh, 2017-11-24:
@jflyper PR is related to selection of wrong DMA configuration, I had the same problem on my board and fixed it quickly, after looking at possible configs. 
@here Enable FEIF interrupt and see if it fires up... I checked DShot 1200 on F7 it looks fine. FYI I was flying with DShot1200 for some time on F4 and never had problems with twitching etc. Perhaps it's related to specific channel configuration?

ronlix, 2017-11-24:
hi,

it is related to the peripheral bridge size and speed or the DMA speed. i noticed on the F3 with dshot1200 the DMA can sometimes not feed the compare register fast enough, so it will repeat its old value.

from what i found, timer channels get feeded after each other (ch1,ch2...) so ch1 of the same timer gets feeded before ch2 and has the best chance to work well. also its seems to be best to use DMA1 and DMA2 in parallel..

this is the reson i never enabled dshot1200 of the KISSFCv1

regards

Felix

ledvinap, 2017-11-24:
Assuming DMA throughput is problem:
- some performance may be gained by using FIFO + different source/destination width
- proshot uses 4 times less edges and minimum pulse width is constrained, it should provide considerably better bandwidth
- using both DMA controllers won't probably help much (they will compete for bus), it may make STM DMA problems worse
- propper register buffering (CCx has shadow register, physical register update may be triggered by different events) may provide some additional margin
- operating FC without considerable margin is quite bad idea ...

conkerkh, 2017-11-24:
- FIFO is used on DShot look at configuration of DMA, however using different width might be a good idea, FIFO should perform data packing/unpacking on it's own according to RM, we would also save RAM
- regarding competing DMA1 and DMA2, arbiter solves those problems according to AN4031

knockoutperformance, 2017-11-24:
sorry about that. fixed that. i did not select the full files in notepad, just what i had on screen.

ronlix, 2018-05-22:
Hi,
sorry that i push this old thing, but im not sure if you fixed it, and i just found out why i dont had that problematic on the kissfc..
see:https://www.rcgroups.com/forums/showpost.php?p=39691310&postcount=4194

my demo of the working code can be downloaded here: https://www.rcgroups.com/forums/showpost.php?p=39688440&postcount=4190

so in fact i think that sometimes the timer increases the DMA CNDTR instandly after it was enabled.. and in this case the first bit gets skipped.. im not sure if my workarround is the best fix, but it seems to be faster then changing timer registers befor starting the DMA channel

regards

felix

DieHertz, 2018-05-22:
@ronlix thank you Felix, we'll look into it soon

haslinghuis, 2022-05-08:
Closing as N/A anymore.

closed 2022-05-08 by haslinghuis