betaflight/betaflight #4629: Support for SPRACINGOSD

nstajduhar, 2017-11-20:
Looking to see if support can be added for the SPRacingF3OSD. I would love to use Betaflight rather than CF but i'm limited currently because there is no firmware for the osd in betaflight

McGiverGim, 2017-11-20:
The SPRACINGF3OSD target is in the code as target, but I can't see it in the releases page. Maybe is a mistake?

mikeller, 2017-11-20:
@nstajduhar: Just flash the cleanflight version of the firmware to the SPRACINGF3OSD board, and use the cleanflight configurator to set it up, then connect it to a flight controller running Betaflight.

@McGiverGim: The SPRACINGF3OSD firmware is not added to the release intentionally - the Betaflight configurator does not support it, and the cleanflight configurator refuses to connect to it, so there is no (convenient) way to configure it. As stated above, the right way to use the board is to flash cleanflight firmware onto it.

nstajduhar, 2017-11-21:
I have tried that when flashing BetaFlight to the FC and Cleanflight to the OSD I get a "Waiting for FC" message on the osd. But when both are flashed with Cleanflight it works perfectly fine and all works well any ideas?

mikeller, 2017-11-21:
@nstajduhar: No, not really. It _should_ work, but it's not really been tested. What version of Betaflight did you use? And did you enable MSP on the port that is connected to the SPRACINGF3OSD?

nstajduhar, 2017-11-23:
@mikeller I tested on latest build of BF at the time along with the latest version of CF for the OSD im going to try again in a few days and hope for the best

nstajduhar, 2017-11-27:
@mikeller Just finished testing using Latest BF and Latest CF on the OSD. Still get the "Waiting for FC" message on the screen tried multiple configurations. MSP is enabled on UART1. Im not sure if I need to supply power to UART1 for it to work maybe or something else. Let me know any ideas you may have

mikeller, 2017-11-27:
Hmm. I haven't tested this in a while. Are both the SPRACINGF3OSD and the flight controller actually powered up?

nstajduhar, 2017-11-27:
They should be, I have the limo going to the PDB/OSD and a 5V jumper going to the FC both of the boards lights turn on and appear to be functioning. And when hooked up to the vtx I can see the OSD and its displaying a message saying waiting for FC so i'm confused why this is happening

nstajduhar, 2017-11-27:
Ill wait I wont be flying with this quad again till the summer anyway so hopefully by then it will be debugged. In the mean time ill try to get a hold of someone at seriously pro and see if they have any ideas

mikeller, 2017-11-27:
That's easy, you can just try to talk to @hydra in here. :-)

hydra, 2017-11-28:
when i get a chance after my relocation to Spain I'll test this.

mikeller, 2017-11-28:
Thanks @hydra. I did some work to pull the configurator and firmware changes into Betaflight to enable the  firmware to read the SPRACINGF3OSD current / voltage sensors, and tested these. I never tested if the SPRACINGF3OSD's OSD works with Betaflight, I assumed it would since it's using the same facilities as MWOSD does (as far as I know).
The changes required to make the Betaflight build for SPRACINGF3OSD work in the configurator were big and conflicted, so I gave up, and pulled the Betaflight build for SPRACINGF3OSD from all releases, to stop users from going down that path only to find out it doesn't work.
If we can't get the cleanflight SPRACINGF3OSD to work with Betaflight, then at least we should put a note in the cleanflight configurator indicating this.

nstajduhar, 2017-11-28:
Thank you @hydra and @mikeller I will stay up to date with this issue and be checking comments frequently. Please notify me of any updates and I am here for testing if you would like any assistance during this process

Mmarsian, 2017-11-30:
I would like to remember you to closed issue #4450. I have it up and running with SPRacing F4EVO and SPRacing F3OSD/PDB. I think that one important thing was to enable MSP and set speed to 115200 on UART1 of the FC.

nstajduhar, 2017-11-30:
MSP and speed is set to 115200. I dont think this is the same issue as that one. Thank you for commenting though @Mmarsian

hydra, 2018-05-05:
@mikeller can we add this to a milestone so stalebot doesn't close it?

mikeller, 2018-05-05:
@hydra: 'Feature Request', 'Bug', and 'Pinned' are labels that can be used to mark an issue / pull request as exempt from being marked as stale.

mikeller, 2018-06-03:
@hydra: Any updates on this? I think there were some fixes to make the SPRACINGF3OSD work in Betaflight.

hydra, 2018-06-03:
@mikeller did you merge the configurator changes required to support it in the end or sites that need doing?

mikeller, 2018-06-03:
@hydra: I wasn't aware that you had opened pull requests for these changes in betaflight-configurator. Let me know the numbers, and I will merge them.

hydra, 2018-06-03:
@mikeller i haven't.  i was asking if you did that already arter reading comments above.

mikeller, 2018-06-04:
@hydra: I have cherry-picked the changes that I could in betaflight/betaflight-configurator#702. But there are parts what's required to support SPRACINGF3OSD in the Betaflight configurator buried in cleanflight/cleanflight-configurator#450, and I couldn't get this to cherry-pick cleanly because of interleaved commits, so it will have to be done in a hand graft.

Since I do not know exactly what is required to get the Betaflight version of the SPRACINGF3OSD firmware supported in Betaflight configurator, I think it's best if you open a pull request for this graft.

closed 2022-05-10 by haslinghuis