betaflight/betaflight #4630: Matek F405 "index not between 1 and 11"

lasharor, 2017-11-20:
I'm trying to remap Motor 6 to softserial 12. Wiped firmware/data to see if this would solve any problems. It did not.

resource motor 5 none
resource motor 6 none
resource serial tx 11 a15 
> Resource is set to A15
resource serial_tx 12 a08
index not between 1 and 11

Diff

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERIAL_TX 11 A15

etracer65, 2017-11-20:
That target is only defined to have one softserial resource - not sure why.

mikeller, 2017-11-20:
@etracer65: I think lack of pins for it, and configurable softserial pins not being a thing when the target definition was submitted, was the reason for this.

lasharor, 2017-11-28:
I don't completely understand why this has been closed. As the F405 target still does not seem to have 2 serial ports defined. Is this something that can't be added?

Motor output 6 is available for remapping.

mikeller, 2017-11-28:
Ah, sorry, the text and title of your issue look like a bug report (for something that is by design), and not like a feature request. Can you please change at least the title?

etracer65, 2017-11-28:
Shouldn't changes like this be the responsibility of the board maintainer? @MATEKSYS

The original addition of the target: https://github.com/betaflight/betaflight/pull/3069

mikeller, 2017-11-28:
@etracer65: Not really. This is a community project, and as such, board maintainers have to live with improvements to their boards being suggested and implemented by the community. It would be different if the suggestion had downsides, but in this case it really doesn't.

lasharor, 2017-11-28:
I've changed the title of this issue and have made it a feature request.

mikeller, 2017-11-28:
Thanks @lasharor.

MATEKSYS, 2017-11-28:
We will try to find some TIM from UART pins. such as TX4/TIM5_CH1, RX4/TIM5_CH2, TX2/TIM9_CH1.
softserial doesn't use DMA.

mikeller, 2017-11-28:
@MATEKSYS: Not sure if a predefined timer is still required for softserial - if I remember correctly, the timer will be taken from whatever pin is reassigned to be a softserial pin.

Oliquatre, 2018-04-10:
Hi, I don't even see the 1st softserial on the config püage - Using 3.3.1

mikeller, 2018-04-10:
@Oliquatre: You need to enable the feature 'SOFTSERIAL' in the 'Configuration' tab for the ports to show up.

Oliquatre, 2018-04-11:
HI,

 

I did already, but issue was it shows up only after the redirection oft he ports, was sure on my old board ot showed up right after enabling.

 

Thx for your support!

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

mikeller, 2018-06-03:
@lasharor: This has been implemented. Can you please close this issue?

closed 2018-06-10 by mikeller