betaflight/betaflight #4633: Omnibus f4

DoctorFPV, 2017-11-21:
http://www.readytoflyquads.com/flip-32-f4-omnibus-nano

looking to put soft serial on this for telemetry.
enable soft serial and it doesn't update in the resources or in the ports.

not alot of info out there so i figured i would try here

etracer65, 2017-11-21:
You need to enable the feature and map the resource to an appropriate pin.  Then it will show up in the serial ports.

https://github.com/betaflight/betaflight/wiki/Single-Wire-Software-Serial

DoctorFPV, 2017-11-21:
The thing is it is never stated that you need to enable the feature in the cli before it will show up in the ports. 

By using resource serial_tx or rx 11 or 12 then the spot on the board then go to ports and you should be able to assign it.

DoctorFPV, 2017-11-21:
It is backwards and confusing and it took me most of the day and my group to figure it out but I got it now. Even Oscar Liang didn't go over that part.

mikeller, 2017-11-21:
@DoctorFPV: You have to understand that resource remapping is still relatively new, and still lacks support in the GUI. Any contributions (of code) towards making resource remapping available in the GUI are welcome!

DoctorFPV, 2017-11-21:
Yeah I'm not mad I just wish when people said you have to enable it, they could be more specific. 

https://oscarliang.com/betaflight-soft-serial/

He just said enable it. Not that you have to assign it a spot then it'll show up.

mikeller, 2017-11-21:
You might have looked at an old version of Oscar Liang's page, the current version is showing how to configure the resource.

jflyper, 2017-11-21:
https://www.rcgroups.com/forums/showthread.php?2989357-OMNIBUS-F4-Nano-Wiring-Worksheet-and-Examples

closed 2017-11-21 by jflyper

DieHertz, 2017-11-21:
>@DoctorFPV: He just said enable it. Not that you have to assign it a spot then it'll show up.
Seems like you did not read carefully, Oscar consulted with me when writing this article and `resource` command and the fact that softserials start at index `11` were mentioned from the very start. It's a recent article, there's no old version.

DoctorFPV, 2017-11-21:
Nah man it could have been more clear but it's all good I got it now

DoctorFPV, 2017-11-21:
You guys can give me shit all you want. Wasn't easy as you think you are kinda biased

wind0r, 2017-11-21:
@DoctorFPV  i think your laste message totaly is not needed here.
Since no one ever said it is easy. @mikeller even said that contribution to make it more easy and available in the gui are welcome and @DieHertz  just said that "Even Oscar Liang didn't go over that part" isnt correct since the blog post talked about resource command in a lot of places.