betaflight/betaflight #4638: Secondary SBUS input MODE, (effectively trainner mode on the quad)

Engineereric, 2017-11-22:
Not really an issue but putting it out there for some help.
I want to add a second SBUS receiver on another UART input.
I will create a new mode that I can assign to say AUX2 on the primary receiver input so that when this mode is activated the first 5 channels on the secondary receiver are used as inputs instead of those form the primary.
Leaving the higher AUX channels as inout form the primary still gives primary authority and so allows secondary control to be switched in and out.
No fancy cables or similar Controllers just two receivers on the quad.
Any thoughts on how to best implement this would be appreciated.
So far my examination shows that a new mode should be easy but there is probably some need to duplicate the SBUS code.

etracer65, 2017-11-22:
This would probably be of extremely limited use for the target audience of Betaflight.

If your goal is redundancy, then maybe you should consider using the new R-XSR receiver from FrSky and pair that with another SBUS receiver (like the X4R-SB or XSR).  The R-XSR accepts a SBUS input from another receiver and manages the redundancy.

Engineereric, 2017-11-22:
Yes it may be limited use (it is a great training tool however). 
My goal is not redundancy.
I really do want to give control over to another receiver input
basically I am now looking at how to duplicate the SBUS code to allow two serial streams.

DanNixon, 2017-11-22:
I really can't see how this is better than the traditional radio-radio trainer mode. That is much less effort to configure as no reconfiguration of FC or additional hardware is required.

The only requirement is to have one radio that can be a master and one that can be a slave (both are pretty common, especially the latter). There is no requirement for both radios to be identical.

martinbudden, 2017-11-22:
@Engineereric , can you explain what you want to use this for?

mikeller, 2017-11-22:
@DanNixon: Does this actually work with two radios of a different brand, running different firmware? (Say for example one a Taranis with OpenTx and the other one a Spektrum.)

DanNixon, 2017-11-22:
@mikeller Yes, I've used my Taranis X9D as a master with both a FlySky i6 and Turnigy 9X as the slave in the past with only a mono patch cable.

mikeller, 2017-11-22:
@DanNixon: Good to know, I did not expect the format of the signal to be compatible.

DanNixon, 2017-11-22:
@mikeller IIRC the PPM signal format is configurable in OpenTX, if not an inverter should be all that is needed. OpenTX lets you do whatever you want with the channels from the slave radio, others may be a bit more restrictive.

richard-scott, 2017-11-22:
The Spektrum outputs a PPM signal on the Trainer Port, that's why the other TX understands it.

ledvinap, 2017-11-22:
I use Taranis as master, Radiolink as slave, with Radiolink SBus receiver in Taranis module bay. Works fine and there is no cable.

etracer65, 2017-11-22:
Yeah, setting up a master/slave trainer connection is pretty straightforward if you have a good "master" radio.  I set it up using my Taranis and an old Flysky radio from a RTF plane so my daughter could learn to fly.  Even made some fancy programming on the Taranis so I could give her control by tapping a momentary switch and then take control back immediately if I moved any of the sticks on the master radio.

Engineereric, 2017-11-24:
I need it for a project I am working on and I could see the benefit to the community as a training aid. Right now I am just hard coding the second SBUS input to UARTR3 and duplicating the SBUS globals, but if the community wanted to use it it would have to be done properly. 
Also means its very easy to plug a second RX in and train some one because face it how many people really have the correct trainer cables laying around?

mikeller, 2017-11-24:
Yeah, I can see value in this. Making SBus (or anything else) reentrant has value in itself, and combining two RX for a trainer mode sounds useful.

The one problem that needs to be solved is 'how to do the switching'. If the entire RX functionality is switched, this will mean that the channel that was used to switch to the trainee also transfers into the control of the trainee. That is clearly not desirable.

martinbudden, 2017-11-24:
@Engineereric , I agree this could be useful and I agree with @mikeller that making the code reentrant has value in itself.  However I am curious: what is the nature of your project and what are you actually planning to use this for?

Engineereric, 2017-11-24:
@martinbudden @mikeller, I'll drop you an email shortly to explain further thanks for your interest and support. P.S. Martin you never mentioned if you got the MidelicF3 Board I sent you.

martinbudden, 2017-11-25:
@Engineereric , my apologies, I thought I had responded about the MidelicF3 board - yes I did receive it, thanks.

PR https://github.com/betaflight/betaflight/pull/4658 makes the SBUS code reentrant. You should be able to create a second SBUS by just duplicating the `sbusInit` function. Let me know if this works for you, or if you have any suggestions for improvement.

Abancalari, 2017-11-29:
Another thought on this feature would be for multiple receivers for redundancy. @etracer65  mentioned FrSky has something like this but it would be  preferable to have a system that works for all types of receivers. One particular case would be to have two Crossfire receivers bound to the same TX and outputting RSSI/ LQ on a channel. The FC then picks which receiver to use based on which has the best signal. Effectively a crossfire diversity scheme with micro receivers instead of the bulk full sized one.

tifoha, 2024-03-11:
Hi! @Engineereric I'm currently working on a pet project for visual navigation, and I was also thinking about a such feature with second sbus receiver(trainer mode). You mentioned, that you have done some solution where you just "hard coded the second SBUS input to UARTR3" 
Is there a way that you can share your fork?

Engineereric, 2024-03-12:
I’m sure it was pushed back into the repository so its there somewhere. There was discussion on its use and implementation Michael Keller had a copy once.I did it because I wanted to give autonomous control to an on board computer.Basically it checked for the state of one of one of the aux switches on the main SBUS input and then depending on the state took the first 4 channels from the second SBUS input.Was quite easy to do because the SBUS code was reentrant.But sorry I don’t have a copy anymore.Sent from my iPhone 8

closed 2026-09-26 by blckmn