betaflight/betaflight #4640: about betaflight max current lit

nyway, 2017-11-22:
Hello
my target is OMNIOUS F4SD
on board current senser，0.5mR Sampling resistance，INA169.
When armed,The OSD display the value of max current is 65.9A，but actually the max current was reach to 100A，everytime OSD dispaly max current 0- 65.9A,no more. We measure the adc port voltage.the voltage  Constantly rising,but  OSD dispaly max current 0- 65.9A.what causes them?

McGiverGim, 2017-11-22:
I don't understand your message. Are you saying that your max current shown is 65,9 but the real is 100 or more? This is a limit of the hardware, you can find more info here for example: https://www.rcgroups.com/forums/showthread.php?2798055-Understanding-Current-Meters

nyway, 2017-11-22:
@McGiverGim  
Thank you for your reply,as you say,my OSD display max current shown is 65.9A,but when i continue grow up throttle ,i measure the adc port voltage was grow up too,but OSD shown the number of current remain unchanged.  hardware circuit should be corrected,because the OSD shown current from 0-65 was correct.but when grow up throttle,no more change

nyway, 2017-11-22:
@McGiverGim
thanks for your help ,I know what you mean

jflyper, 2017-11-24:
@nyway 
You can increase the range by stacking another shunt resistor onto the existing one.
https://www.youtube.com/watch?v=mdH64v_ZYG8

closed 2017-11-24 by jflyper