betaflight/betaflight #4641: question "FPV angle mix"

adrianmiriuta, 2017-11-22:
@borisbstyle 
BOXFPVANGLEMIX is intended to make yaw rotation more intuitive when flying FPV

if the quad flying angle is not exactly the same with the preset  camera angle 
than the BOXFPVANGLEMIX correction will be wrong ! 

I think it would be better to transpose yaw rotation  in EarthFrame
so that yaw movement would be earth parallel for any flying angle . 

what are your thoughts about this ?

etracer65, 2017-11-22:
The feature is designed to relative to the FPV camera's frame of reference - hence the name "FPV angle mix" and works the way it should.  This is not a line-of-sight feature.  If you want yaws to remain parallel to the ground then use angle or horizon mode with auto-leveling.

adrianmiriuta, 2017-11-22:
@etracer65 
sorry I think you misunderstood me
I do not want  sensor frame (copter frame) to be parallel to earth frame  (aka angle or horizon mode).
I men a true 3D earth frame parallel rotation, 
so that roll and pitch angle remain unchanged  (copter frame ).

etracer65, 2017-11-22:
The purpose of the FPV angle mix is to automatically cross-coordinate yaw when rolling to compensate for camera uptilt.  The effect being that rolls will "look" centered from the FPV view.  The underlying design was to make the experience more airplane-like while flying.  It does what it is supposed to do.  If you want some other capability then document and request that rather than as some correction to the existing feature (which is working correctly).

adrianmiriuta, 2017-11-22:
@etracer65  I do not want to request a feature
it is a question !

say you have set a camera angle 35 deg and flying 35 deg pitch 
when you yaw it is a earth frame parallel rotation.
If you are flying faster say with 45 deg pitch than your yaw will not be earth parallel  
it will change roll and pitch angles and you will loose altitude.
a 3D rotation would preserve roll and pitch angles  and make it intuitive.

It a yaw roll compensation without a fixed angle.

etracer65, 2017-11-22:
You're still not understanding the purpose of FPV angle mix...  The feature is designed to compensate for the camera up-tilt to make it appear ***from the FPV pilot's perspective*** that the requested rotation around each axis is as if the camera up-tilt was zero degrees.  The FPV angle mix performs the cross-coordination on the other axes so that the pilot doesn't have to do it.  The frame of reference is the vector defined by the FPV camera angle.  If you still don't understand, hold a quad with the up-tilted camera level to the ground.  The frame will obviously be tilted forward with negative pitch.  Not yaw around the center of the flight controller's frame of reference and see what happens to roll and pitch.  Normally the pilot compensates for this in flight.  The FPV angle mix does it instead.  All that is needed to calculate this is the angle of the camera relative to the flight controller.

> say you have set a camera angle 35 deg and flying 35 deg pitch
> when you yaw it is a earth frame parallel rotation.
> If you are flying faster say with 45 deg pitch than your yaw will not be earth [...]

What you are describing is already what will happen in angle or horizon mode.  When you yaw the flight controller will maintain the current pitch and roll regardless of what they are (camera up-tilt is irrelevant).  To maintain the orientation of the other axes the flight controller must have a frame of reference.  In this case it's provided by the accelerometer and the orientation to the earth.

adrianmiriuta, 2017-11-23:
@etracer65 
I think i understand how "FPV angle mix works".

My question was  
Is a fixed camera angle better as a dynamic one ?

Taking in consideration that when you are  flying 
you do not hold your pitch angle  all time at the same value with the cam angle.

DanNixon, 2017-11-23:
@adrianmiriuta 

>  I do not want to request a feature
it is a question !

So what *is* your question?

What you have done so far is suggest a feature.

adrianmiriuta, 2017-11-23:
@DanNixon 
---
My question was
Is a fixed camera angle better as a dynamic one ?
---
I am a beginner in FPV and not sure what users expect !

McGiverGim, 2017-11-23:
If you make a "dynamic one" you limit the kind of movements that your quad can do. It will always make the yaw referenced to the earth, and you don't always want it. Making it reference the quad or the cam, you can do more movements in the 360 degrees space. 

So NO, is not better a dynamic one.

adrianmiriuta, 2017-11-23:
@McGiverGim 
that assumption is wrong !
there is no movement limitation.

The difference is the reference system,
with dynamic angle correction 
yaw movements are transposed from copter bound reference to earth bound reference.

You have the same liberty degree .

DanNixon, 2017-11-23:
If yaw is relative to earth and pitch and roll are relative to the quad then you do have situations of reduced movement (for instance when the quad is facing straight down).

You also have the yaw axis appearing reversed when the quad is inverted.

McGiverGim, 2017-11-23:
Exactly, if your quad is face down, for example, now moving the yaw you can make it to "yaw" until the face is, for example, to the sky. If you make it dynamic you cannot make it using this axis, you can do it using the pitch but the quad will be reversed too,if your reverse it with roll the end will be the same, but not the movement line.
If you plan only to fly "equilibrated" (like angle mode but using acro) maybe there's not too much difference, but for this is better to use angle/horizon. If you plan to fly really acro, with freestyle movements... the difference is clean and dynamic one is not a way to go.

adrianmiriuta, 2017-11-23:
@DanNixon 
@McGiverGim 
yes You are correct, I agree.
Thx

closed 2017-11-23 by adrianmiriuta