betaflight/betaflight #4646: OSD Feature Request

BludzUK, 2017-11-23:
Hi,

I'm using ESC telemetry from _32 ESC's and really like the extra data. The ESC temperature is nice to see but it being an average of the 4 esc's is not that helpful. I can only see one reason to have the temperature displayed and that's to see if they are getting too hot. Problem is that when 1 esc is overheating and the others are fine, the average temp doesn't increase dramatically.

Not sure how easy it would be to add but an ESC warning would be very useful. For example, 1 ESC goes above a set temperature, the osd displays the esc number so you know which has a problem.

Keep up the great work and hope this is the right place to post this idea.

jflyper, 2017-11-24:
Duplicate of #4584

jflyper, 2017-11-24:
Sorry, this one is for temperature.

mikeller, 2017-11-24:
@BludzUK: It's not the average (which would be not very useful indeed), it is the temperature of the hottest of the ESCs, so that half of your request has already been solved.

BludzUK, 2017-11-24:
Ah right, that makes more sense. Was I wrong to think that the voltage is an average of the 4 esc's? That's why I presumed the temp was the same.

Good to know that the osd/fw is allready doing part of what this feature would require. Perhaps not such a good feature now I know what it's doing.

Thanks for the info if nothing else.

mikeller, 2017-11-24:
No you weren't wrong. Voltage and RPM are averages, current and consumption are sums, and temperature is max. They are the aggregation functions that were thought to give the most useful values for pilots. (And your feature request seems to be confirming this. :-) )

So you can already get the temperature of the hottest ESC shown in the OSD. I've modified the title to ask for a warning based on it to be added, which should be relatively simple.

BludzUK, 2017-11-25:
Thanks for all the details, looking forward to maybe seeing this added.

Perhaps a warning for a big difference between esc's rather than just a set temp to warn at. So it warn if there is more than 20-30 degrees C between any 4 esc's. This would alert you of a potential problem.

1337flite, 2017-11-27:
I would think you might want to warn on an absolute value or a difference.
Otherwisei if all your ESCs are quite hot you're your alarm might not trigger even though all the ESCs are hotter  than you'd like them.

BludzUK, 2017-11-28:
Good point 1337flite

I was thinking on this further and feel that perhaps it would be possible to actually have the ESC telemetry displayed like some other osd's. Where you have each motor/esc data in it's corresponding corner of the screen. So motor/esc 1 in bottom right, 2 in top right and so on. Then the individual readings for each, with individual warnings. Having the option to do this would be cool, but perhaps refining the warnings on the current state would be just as useful.

BludzUK, 2017-11-28:
Has anyone requested the feature to have 'Anti-Gravity' show up in the osd when it's active? Would be great to actually see if it's working or not, and whether the threshhold or other parameters need adjusting.

mikeller, 2017-11-28:
@BludzUK: A comment is not a good place to ask for a new feature. Please open a separate issue and we can track it as a feature request.

BludzUK, 2017-11-28:
Fair enough, just didn't want to open a new one if it had been requested allready. On it though, thanks

pulquero, 2018-03-30:
@BludzUK I've just done a PR to add an OSD warning if motors fail (rpm==0). I could probably do something similar for temperature, e.g. "TMP1 4" if 1 and 4 overheat but 2, 3 are ok. The rpm warning should be changed from "ESC1  4" to "RPM1  4". If we are going to have a cli option to set overheat temperature, likewise we can also have an option for min rpm.

BludzUK, 2018-03-31:
@pulquero That's great, I was hoping this might get implemented soon. What is the reason for an osd warning of a motor failing? You would soon know about it. Maybe to help solve desync issues? Still sounds like you can indeed use a similar method for the temps.

I thought of another useful warning that needs to pull data to the osd. CPU load. This would be great now we can push F4's.

Thanks a lot for offering to add my feature request. I'll have to keep an eye on the nightly builds for it.

pulquero, 2018-03-31:
Idea is so you would know which motor(s) had failed or in fact if a motor had failed at all and the cause was something else.

mikeller, 2018-03-31:
@BludzUK: For ESC temperature warnings see #5583.

The problem with CPU load is that calculating it takes up resources, and is therefore only done when in CLI.

BludzUK, 2018-03-31:
@mikeller That's a shame but don't want to add to the problem by using more resources. The code for adding the temps is really helpful for if I find some time to play with the code myself, this will tell me partly how to pull data to the osd.

@pulquero I'm still struggling to imagine where this would be useful as it's pretty obvious if a motor isn't working, and not hard to tell which one it is. I'm sure it has a use, but can't help but think that you will know it has failed before the osd even warns you. Also only going to tell you that an esc isn't making a motor spin, but not that a motor has failed, could be an esc or FC failure.

pulquero, 2018-03-31:
@BludzUK I don't find it easy to tell which it is. Out of the blue you find yourself in a spiral of death and then before you know it you are in the ground. You re-arm and everything looks to be working fine. I'm fortunate enough to have goggles with dvr, but the quality doesn't make it easy to determine the direction of rotation. Sorry, I meant to say in conjunction with a blackbox log you could better determine the cause.

BludzUK, 2018-03-31:
@pulquero I get it now. I usually run a hd cam and can slow it down if I don't have blackbox. Been a long time since I had an esc fail and then work fine after, but it does happen. Totally get it now. Thaks :)

pulquero, 2018-05-05:
This is now in master and can be closed.