betaflight/betaflight #4648: Hobbywing f4 led's not working

Xtopher98, 2017-11-24:
Hello! I just set up a new Hobbywing f4 and on the stock betaflight version (not sure what version that was) the less worked just fine. When I flashed 3.2.2, everything else worked but the led's would not turn on. I know its not the physical connections because I tested all the wiring with a multimeter after it stoped working. Not sure whats wrong.

This is using the OMNIBUSF4SD firmware.

jflyper, 2017-11-24:
@Xtopher98 
Try
```
resource led_strip a8
```

If it doesn't work, please consult your dealer.

Ref:
https://github.com/betaflight/betaflight/wiki/Board---OMNIBUSF4#v317-and-later-led-strip-pin-assignment-has-changed-for-omnibusf4sd-target

closed 2017-11-24 by jflyper

mikeller, 2017-11-24:
@Xtopher98: For a board to be supported in Betaflight, the board manufacturer has to either develop and submit a target definition for their board, or else at least seek a dialogue with the Betaflight community, and provide schematics and sample boards, so that somebody from the community can contribute a target definition for this board. Please contact the manufacturer and / or your supplier and ask them to do any of the above so that support for your board can be added to Betaflight.

Xtopher98, 2017-11-24:
Remaping led_strip to A8 did the trick! Thanks for your help!