betaflight/betaflight #4652: OSD should show "up" in turtle mode

IB1387, 2017-11-24:
Hi guys, a little feature request here:
It would be nice to have the OSD give a recommendation which way is up when switching on flip over after crash mode using the accelerometre. Sometimes it's hard to guess which direction one should try to get the quad to flip over only using the fpv feed. 
Thank you all for this great piece of software.

Best regards,
1387

Vidalcris, 2017-11-24:
Maybe not really a recommandation but personnaly the switch for turtle mode also set level mode so when i arm the quad do the job itself ;)
I just need to arm for 1sec and the job is done !

mikeller, 2017-11-25:
@Vidalcris: That actually works? That's quite a good 'hack'! :-D

Vidalcris, 2017-11-25:
Yes i use this method on my 2inch quads, that work really nice ;-)

IB1387, 2017-11-25:
Nice Trick, thanks, I'll try that. But it could lead to problems when one side is stuck somewhere. So my feature request still holds. :)

kmitchel, 2017-11-26:
@Vidalcris I gave this a try...didn't seem to work.

IB1387, 2017-11-26:
Just tried it aswell with no success. I suspect that only works with the "old type" turtle mode where all motors were spinning when turtle-mode was active and the copter was armed?

kmitchel, 2017-11-26:
I think it 'limits' turtle mode...see if you can get it to flip over 360.

kmitchel, 2017-11-26:
Looks like turtle is using RC deflection...so it isn't doing anything.

Vidalcris, 2017-11-26:
Mhm yeah sorry i just tried with 3.2.2 and this is not working anymore :-/
I think that was working with 3.1.7..
For me the newest method is not working really nice .. i mean its way harder to return the quad now.

jflyper, 2017-11-26:
"_Random Shake_" deturtle?

kmitchel, 2017-11-26:
This could be a new feature tho...if the accelerometer is enabled, then
limit tutle mode appropriately.  Kind of an assisted turtle mode, not fully
automatic.

Vidalcris, 2017-11-26:
@kmitchel its a really good idea in my opinion ;)

etracer65, 2017-11-27:
Back to the original request...  I think it would be quite useful to have a "turtle arrow" that indicates the direction to push the stick based on the "lowest" side.  Since the quad is inverted the "lowest" side is actually the "highest" side as far as the accelerometer is concerned.  This could give you the best chance to determine which props are not stuck.

IB1387, 2017-11-27:
@etracer65 Thank you. Using the accelerometer to not let the quad do a 360 flip and landing on its head again could be usefull too, but I'd really like the arrow. I don't really see the need for automatic flip over for normal 5" quads on grass though. That's maybe more for tiny whoops or so.

kicekilla, 2017-12-03:
Following this, the arrow would or direction suggestion would be nice. there have been a few times when I just have to try a few different ways.. scary bc I don't want to burn an esc/motor. 

It would also be nice to be able to enable acc to limit the quad flipping as sometimes I'll flip back over and land right back upside down.

mikeller, 2017-12-03:
@kicekilla: Have a look at 'Pitch Angle' and 'Roll Angle' OSD elements. Checking if they are positive or negative will allow you to determine if you need to move forward / right or backwards / left.

IB1387, 2017-12-03:
@mikeller Thank you for the suggestion. Having that would certainly help but I think there is still value in having a "proper" turtle indicator, as I don't want to clutter my OSD with that information all the time other than when I'm in turtle mode. Secondly, having to look at values and going from there what the highest side is, would be more difficult than just seeing an arrow and turteling in that direction.

etracer65, 2018-08-27:
Closed via #6467

closed 2018-08-27 by etracer65