betaflight/betaflight #4654: Motor Stall on Punch out BF 3.2.2 and BLHeli_32 V32.1

dgrigg123, 2017-11-24:
Had an event today on a punch out using a 5S battery.  I would upload a blackbox log but didn't see a way to show pict or upload a file in this forum post.
Anyway, straight up punch out shows motor 1 going to 100%, but in fact it simply stalled/stopped.  Other 3 motors responded correctly.  Dropping throttle got motor 1 running again, and luckily I was high enough to recover.  I tried to reproduce 1 or 2 ties again, but did not on 4S batteries and I was a bit afraid to try too much again on my 5S.  I've never had this happen before but wondering if there is a known issue with this in the latest builds.

Diff is shown below.  I have all the latest BF and BLHeli_32 firmware.
My FC is the CLRacing F4S.
Current Limit set to 35A in BLHeli.
Wraith32 with BLHeili_32 V32.1
 Let me know if you have any thoughts on this and if a blackbox log would help and how to send.
Thank you,

\\\# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.2 Nov 11 2017 / 00:24:23 (77782e602) MSP API: 1.36

# name

# resources
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 7 1700 2100
aux 1 26 1 1700 2100
aux 2 27 5 1700 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_trim_pitch = 4
set acc_trim_roll = 4
set fpv_mix_degrees = 25
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT1200
set failsafe_kill_switch = ON
set align_board_roll = 180
set align_board_pitch = 180
set pid_process_denom = 1
set osd_vbat_pos = 2071
set osd_rssi_pos = 2113
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2081
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2166
set osd_mah_drawn_pos = 2134
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 2342
set osd_pid_pitch_pos = 2374
set osd_pid_yaw_pos = 2406
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2103
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2250
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_tim_2 = OFF
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 4000
set dterm_setpoint_weight = 100
set p_pitch = 80
set p_roll = 75
set p_yaw = 90
set i_yaw = 50

# rateprofile
rateprofile 0

set rc_rate = 155
set rc_rate_yaw = 155
set rc_expo = 20
set rc_expo_yaw = 10
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75

#

etracer65, 2017-11-25:
I saw your post on rcgroups - it looks like a classic momentary de-sync.  Try increasing the digital idle or possibly play with some of the parameters in BLHeli_32 like de-mag compensation.  If the FC was commanding throttle and the ESC/motor/prop was not producing thrust then the fault is not with the FC or Betaflight.  Since the problem manifested while using %S and was ok on 4S the extra stress of %S may require some tweaking of the BLHeli_32 parameters.  I would recommend posting in the BLHeli_32 rcgroups or maybe a group for your brand of ESC.

dgrigg123, 2017-11-25:
etracer65,  Thanks.  I will move to RCGroups with the issue.  I'll maybe try both increasing the Motor Throttle Idle Value and investigate the de-mag compensation a bit and see if it reproduces.  Thank you.

Fuzzball27, 2017-11-28:
Just my 2 cents after having been through something similar: check your B_Heli configuration (the chrome app "B_Heli Configurator" is great for this). Specifically, check that your motor timing matches the manufacturer's recomendation. Most people just blindly run medium-high timing because they assume bigger is better, but motor timing is nothing to do with "speed" and everything to do with 3-phase timing and EMF.
Case and point: if the reccomended timing is 0 deg then do low timing; if it's 5-10, then do med-low timing; etc, etc.

dgrigg123, 2017-11-28:
Thanks Fuzzball27, but I have yet to see any manufacture recommendations for timing on my motors, particularly the RMRC 2206 2550kV motors.  Let me know if you have some recommendations on how to find that.  Also, I think I have to use the BLHeli_32 suite application since I haven't heard of a chrome app for BLHeli_32 being out yet.
However, I did decide to our the timing to "auto" at the potential expense of some possible top end power.  So far the event has not reproduced, so maybe letting BLHeli monitor and adjust my timing has helped.
Thanks again.