betaflight/betaflight #4661: Remapping to motor 8 on omnibus f3

Bugg69s, 2017-11-25:
Hi, today I wanted to make my quad cleaner. But I was really smart and cut esc signal wire too short -_- . I needed to move motor 4 to motor 8 pad. I did it. And then I opened BF and wanted to change motor 4 to motor 8 pad. But I couldn't find motor 8 in configuration or Google :/  is there way to remap motor 4 to motor 8pad ? (pwm4 to pwm8)

mikeller, 2017-11-25:
https://github.com/betaflight/betaflight/wiki/Remapping-Motors-with-Resource-Command-(3.1)

closed 2017-11-25 by mikeller

Bugg69s, 2017-11-25:
@mikeller what's the motor 8 code ? (Like a B06,A03, etc...) I will need that code. I can't remap anything to that motor 8pad without that code

mikeller, 2017-11-25:
Well if it's not shown in the output of `resource list` then I guess your flight controller does not define a motor 8. Please read the documentation that is available before asking questions.

jflyper, 2017-11-26:
@Bugg69s 
PWM7 is PB7
PWM8 is PB6
You can find the info here.
https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUS/target.c

jflyper, 2017-11-26:
I'm closing this, as this is a support request.