betaflight/betaflight #4662: Dshot broken on NeroF7

goodnerdboy, 2017-11-25:
Flashed the latest build [#1843](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1843/) on a NeroF7 and when i set to Dshot600 or Dshot1200 only motors 1 and 2 work.
Motor 3 and 4 don't spin at all, even from the motor tab in BFC.

mikeller, 2017-11-25:
This is potentially caused by #4073. Do you know what the build that you used before this one and that had Dshot still working was?

MJ666, 2017-11-26:
I have also problems with motor 3 and the AlienFlightNG F7. Have not checked for more than four motors yet. More motors may be affected. Motor 3 is using an inverted timer, not sure if this is related. Commeting USE_DSHOT_DMA is fixing this problem.

sambas, 2017-11-26:
Tested latest master on anyfcf7, no issues detected upto octo setup. Mixed timers but no N-channel on this. Can you guys dump 'resource' and 'resource list'?

goodnerdboy, 2017-11-26:
I think there is a DMA conflict with LEDs. If i disable LED_STRIP Dshot works like a charm, if i enable it, motor 3 and 4 don't spin at all.
When led_strip is active, DMA1 Stream 7 is used by LED_STRIP, and none by motor 4
When led_strip is not active, DMA1 Stream 7 is used by motor 4 :

`
Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: LED_STRIP
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
`

`
Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
`

sambas, 2017-11-26:
Yes exactly. for Nero you can try this, https://www.dropbox.com/s/35u1uk26sxeo4bv/betaflight_3.3.0_NERO.hex?dl=0

goodnerdboy, 2017-11-26:
@sambas it seems to work, at least on the bench.
I’ll fly this afternoon to confirm :)
Thanks

MJ666, 2017-11-26:
Here is the output for the AlienFlightNG F7 with 8 motor outputs activated and DSHOT600.

DMAR disabled:

`# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 5
A01: MOTOR 7
A02: FREE
A03: SERIAL_RX 2
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: BARO_CS
B00: MOTOR 4
B01: FREE
B02: RX_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SDCARD_CS
B11: SDCARD_DETECT
B12: FLASH_CS
B13: SPI_SCK 2
B14: MOTOR 3
B15: FREE
C00: FREE
C01: FREE
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: FREE
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 6
C09: MOTOR 8
C10: FREE
C11: FREE
C12: LED 1
C13: BEEPER
C14: MPU_EXTI
C15: COMPASS_CS
D02: LED 2

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 5
DMA1 Stream 3: FREE
DMA1 Stream 4: MOTOR 7
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 1
DMA2 Stream 3: MOTOR 3
DMA2 Stream 4: MOTOR 6
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 8
`

DMAR enabled:

`# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 5
A01: MOTOR 7
A02: FREE
A03: SERIAL_RX 2
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: BARO_CS
B00: MOTOR 4
B01: FREE
B02: RX_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SDCARD_CS
B11: SDCARD_DETECT
B12: FLASH_CS
B13: SPI_SCK 2
B14: MOTOR 3
B15: FREE
C00: FREE
C01: FREE
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: FREE
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 6
C09: MOTOR 8
C10: FREE
C11: FREE
C12: LED 1
C13: BEEPER
C14: MPU_EXTI
C15: COMPASS_CS
D02: LED 2

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 7
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 4
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: MOTOR 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
`

jflyper, 2017-11-26:
@MJ666 I think three DMA stream usages corresponds to TIM3_UP (1,2), TIM5_UP (1,0) and TIM8_UP (2,1).

sambas, 2017-11-26:
@MJ666 do you have scope or logic analyzer to see if N-channel output has any signals coming out of it?

sambas, 2017-11-26:
@MJ666 try this https://www.dropbox.com/s/qsv1wgqm3216sej/betaflight_3.3.0_ALIENFLIGHTNGF7.hex?dl=0

MJ666, 2017-11-26:
There is no signal at all for the motor 3 output. I will try your hex when back home later today.

MJ666, 2017-11-26:
@sambas Have tested your hex build. Here all 8 motor outputs lock good on the scope. Can not fly this at the moment.
[image]

jflyper, 2017-11-26:
@sambas If output channels are on different timers, they are grouped into corresponding TIMx_UPs, right?

sambas, 2017-11-26:
yes. Resource list only shows last initialized output for DMA channel.

jflyper, 2017-11-26:
Yeah, that's what I thought... We may create a new owner for UPs to avoid confusion. `TIMUP`?

goodnerdboy, 2017-11-26:
@sambas my Nero flew great this afternoon, thanks for the fix!

jflyper, 2017-11-27:
Closed via #4664

sambas, 2017-11-27:
Actually not fixed with #4664 that was only for n-channel fix @MJ666 had. Target fix not included.

jflyper, 2017-11-28:
Are you going to submit a NERO target PR? Or, is this problem (collision between TIMUP and other devices) should investigated in a broader context (like for all F7 targets)?

sambas, 2017-11-28:
I can do it but target maintainers should go thru them.

mikeller, 2017-11-28:
Thanks, @sambas!

lexusburn, 2018-02-17:
@sambas it seems that i have the same problem with OMNIBUSF7 (V1).

There's a hardware issue with the original LED_STRIP pinpout on V1 and V2. Because of this problem i can't use the original pinout to get LEDs working correctly. Now i've use the pinout for RX6 and remap it in betaflight CLI with these commands:
`
# remap led strip: switch RX6 with LED_STRIP
resource SERIAL_RX 6 E01
resource LED_STRIP 1 C07
resource SERIAL_RX 6 D12
`
With these i had led strip and dhsot working on BF 3.2.?
Beginning with aprox 3.2.5 i've got problems with DSHOT. Motor 3 will not spin on shot anymore.

If i don't do it like this i have these DMA:
`
# dma

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: MOTOR 3
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
`

After doing this i get these DMA with BF V3.3.0 RC1:
`
# dma

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: LED_STRIP
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
`

Can you help me to get both working again. How can i prevent that LED_STRIP collides with MOTOR 3 DMA?

mikeller, 2018-02-17:
@lexusburn: @jflyper might be able to help.

jflyper, 2018-02-18:
I'll look into this tonight.

lexusburn, 2018-02-18:
@mikeller: thank you! 

@jflyper: there's a level converter between processor pin and pinout. Because of this, the remapping to another pin is needed to get led strip working, but it breaks dshot because of a dma conflict. Thank you for your investigation! :-)

jflyper, 2018-02-18:
@lexusburn 
I'm aware of the hardware issue that causes the default LED strip output pad not working.
The best workaround is to apply the mod described in https://github.com/betaflight/betaflight/issues/4429#issuecomment-339552260, which I believe you are aware of.

I understand that you introduced the workaround to assign RX6 as a simple alternative LED strip output, but unfortunately, master code repository can not deal with random changes that users make.

That being said, you can try followings with 3.3.
1. Use burst mode Dshot.
`set dshot_burst = ON`

2. If above does not work, you may want to try TX3 (PB10) instead. If you are using UART3 for something else, you have to rewire it to UART6, of course.

In the future, uses will have more flexibility on resource assignment including DMA.

lexusburn, 2018-02-18:
@jflyper: setting burst mode on is working for my problem. :-)
Many thanks for this quick help. 👍  
Can you tell me what burst mode means and why this fixes my problem?

closed 2018-04-21 by DieHertz