betaflight/betaflight #4663: [Feature request] Use barometer for landing phase during failsafe

Saxin, 2017-11-26:
Please consider adding option to land using pre-configured descent rate vs. throttle value to allow landing the copter in case of a failsafe event. (Many FC's today come with a baro, it would be nice to utilize it)

jflyper, 2017-11-26:
There would be a quite a long road map to go, and even if completed, it will be an opportunistic feature.

- Altitude estimation and control (or hold atm) hasn't been maintained for a long time, probably since BF was forked from CF, and many issues were reported here. There even is a recent report that ALTHOLD when engaged, flip the copter over (although it hasn't been confirmed).

- Barometer code, too, is not really accurate nor reliable to be used in an autonomous function. There has been a report that measurement fluctuates, sometimes at a degree of kilometers.

Do iNav have this feature? I think it is a feature that would be implemented in iNav, then ported to BF if it is proven to work safely, consistently and as intended.

digitalentity, 2017-11-26:
Barometer-controlled landing in INAV is not implemented. There is an emergency landing sequence, but it's not linked to failsafe yet.

Saxin, 2017-11-26:
@jflyper  - I know the general logic of Inav - BF porting, yet this feature can save people a lot of $$$ and implementation is relatively straight forward - I'm suggesting something complicated nor by any means alt hold framework and such.

In failsafe FC will force channel to angle mode - this works well as of now.
Use baro readings, probably filtered to trow out bad samples to descent, throttle values can be capped at some reasonable values.
Detect ground meeting and disarm.

digitalentity, 2017-11-26:
Limited baro-based control should be possible. In BF we don't need precision altitude/speed control, just limit rate of descent to some sane value. I might give it a try if I have the time

Saxin, 2017-11-26:
@digitalentity If I can help with something - please let me know ?

knockoutperformance, 2017-11-29:
could a lidar unit be used for this also?

jflyper, 2017-11-29:
I'm expecting it to behave better in vibrating environment. We'll see.

Thomsen76, 2017-12-16:
Will there still be a problem with  lack of CPU power, on F7 boards?.

closed 2026-09-26 by blckmn