betaflight/betaflight #4673: Yaw RC Command Runaway on Punchout

Fuzzball27, 2017-11-27:
I am having an issue where the yaw RC Command jitters on a throttle punchout despite no yaw input from the sticks. On rare occasions, this jitter turns into a full blown endpoint command where the yaw RC Command is 2000 or 1000, despite the stick being at center yaw position. This is definitely a software issue; I have attached a BB screenshot of one such occurrance. Take note that the stick yaw input is centered, but the RC Command is full biased to 2000.

I am suspicious that this issue may be related to the rxrange endpoint adjustment algorithm, but only because that's the only parameter relating to stick inputs that I've adjusted from the default.

This problem has become immobilizing and it is a real bummer.

[image]

```
# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.2.2 Nov 11 2017 / 00:35:15 (77782e602) MSP API: 1.36

# name
name GLDN BLLT

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature AIRMODE
feature DYNAMIC_FILTER

# beeper

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100
aux 1 13 2 1800 2100
aux 2 35 1 1800 2100

# adjrange

# rxrange
rxrange 0 1067 1915
rxrange 1 1067 1915
rxrange 2 1067 1905
rxrange 3 1071 1916

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set gyro_use_32khz = ON
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rx_min_usec = 900
set rx_max_usec = 2100
set dshot_idle_value = 100
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 1200
set failsafe_kill_switch = ON
set failsafe_procedure = AUTO-LAND
set align_board_yaw = 270
set bat_capacity = 1300
set vbat_max_cell_voltage = 44
set vbat_scale = 109
set ibata_scale = 180
set beeper_dshot_beacon_tone = 1
set osd_units = IMPERIAL
set osd_cap_alarm = 1040
set osd_tim1 = 2561
set osd_vbat_pos = 2049
set osd_rssi_pos = 40
set osd_tim_1_pos = 2071
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_vtx_channel_pos = 377
set osd_current_pos = 385
set osd_mah_drawn_pos = 2082
set osd_craft_name_pos = 2058
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2346
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_tim_2 = OFF

# profile
profile 0

set vbat_pid_gain = ON
set anti_gravity_threshold = 240
set anti_gravity_gain = 2000
set p_pitch = 35
set i_pitch = 90
set d_pitch = 13
set p_roll = 25
set i_roll = 90
set d_roll = 9
set p_yaw = 20
set i_yaw = 40
set d_yaw = 10

# rateprofile
rateprofile 0

set rc_expo = 10
set thr_mid = 23
set thr_expo = 15
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 72

#
```

etracer65, 2017-11-28:
Please upload the blackbox log for analysis.  From the screenshot I'm seeing a gyro failure on the yaw axis as there's no way the yaw rate could possibly oscillate that quickly between the extremes.  Of possible importance - this flight controller has the ICM-20602 gyro which is known to have overflow related issues.  Since the yaw was static before the event it seems unlikely to be this as it requires yaw rates in excess of 2000 deg/sec to trigger the overflow.  After the initial oscillation it appears it may have gone into overflow when it locks at 1980 deg/sec.  The PID controller is pushing back to stop the yaw but if the gyro is inverted in overflow then the PID controller will reinforce the rotation since it's getting invalid gyro data.  Still looks like a failure in the gyro though - particularly since you're seeing occasional yaw jitter or oscillation.  That's typical of gyro problems.  The ICM-206xx series and its ancestor the MPU6500 are known to be very sensitive to vibration.  If you don't have the flight controller soft-mounted to control you may have to do this to make this flight controller usable.  It's possible some extreme jitter was enough to trigger the gyro overflow leading to the runaway.

Fuzzball27, 2017-11-28:
That makes much more sense than what I was suggesting. Thank you for the input.
I do have soft mounts, and I just had the epiphany that this issue arose after I did a firmware update, so I may just revert for the time being. I'll repost in this thread if the issue persists after reverting.

Fuzzball27, 2017-11-28:
Here are the BB logs. The first is the one pictured above. At the end of the 1st log, I was about 100ft above the ground, so I disarmed to prevent my quad going to the moon. The second log is an attempt to slow the fall as I re-armed the quad about 30ft above the ground.
I added the .txt at the end of the file name so that github would allow me to upload the files.
[runawayQuad.BFL.txt](https://github.com/betaflight/betaflight/files/1508301/runawayQuad.BFL.txt)
[fallRecovery.BFL.txt](https://github.com/betaflight/betaflight/files/1508300/fallRecovery.BFL.txt)

Fuzzball27, 2017-11-28:
It looks like this was addressed a while ago here: https://github.com/betaflight/betaflight/issues/3912#issuecomment-347419895

closed 2017-11-28 by Fuzzball27