betaflight/betaflight #4676: LUXF4OSD target in BF 3.2.2 not working for Racerstar TattooF4S Flight Controller

markusbrand, 2017-11-28:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
When flashing the Racerstar TattooF4S Flight Controller with BF 3.2.2 (target LUXF4OSD) the ESCs and Buzzer are not working
2.) Include ways to reproduce the problem;
Flash BF 3.2.2 with target LUXF4OSD to Racerstar TattooF4S
Go to motors tab - try to spin the motors -> doesn't work
Go to configuration tab - change to DSHOT1200 - go to motors tab -> still not working
Disconnect Betaflight
Open BLHEli 32 Suite - try to read ESC/try to Flash ESC/try to spin motors -> nothing working

Flash BF 3.2.1 with target LUXF4OSD to Racerstar TattooF4S -> everything is working properly (even flashing new BLHeli version through passthrough)
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
https://www.banggood.com/Racerstar-TattooF4S-30A-BLHELI_32-4in1-ESC-5V-BEC-w-F4-Flight-Controller-AIO-OSD-Current-Sensor-p-1184869.html?rmmds=search&cur_warehouse=CN
wired motors accordingly - soldered buzzer to pads
added battery lead
camera/vtx not installed yet
4.) Add the used configuration and firmware version;
Betaflight 3.2.2 target LUXF4OSD
protocoll DSHOT 1200

jflyper, 2017-11-28:
@markusbrand 
We are sorry, the flight controller is not supported.
Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2017-11-28 by jflyper

jflyper, 2017-11-28:
@markusbrand 
FYI, local test performed with Omnibus F4 AIO (gen1), which is identical to LUXF4OSD except for the flash chip configuration works without an issue with 3.2.2.