betaflight/betaflight #4687: Feature Request - Anti-Gravity icon displays in osd when active.

BludzUK, 2017-11-28:
Hi,

Can we get an icon or some text displaying in the osd when 'Anti-Gravity' is active? So we know when our stick commands are activating it and whether it needs adjusting.

Thanks

SRWieZ, 2017-11-28:
Like ESP light on car dashboard, great idea!

JyeSmith, 2017-11-28:
Could this be added as an alarm e.g. like when crash recovery is activated?

McGiverGim, 2017-11-29:
I think that added as alarm is not a good idea. The anti-crash is an emergency feature, but the anti-gravity isn't. It will be executed a lot of times every flight and the message appearing like alarm in the middle of the screen can be distracting.
I think is better that it appear in a non distracting place. Maybe change by one second the "flight mode"?

mikeller, 2017-11-29:
Alarms are configurable by type, so will be up to users to disable this particular one if they find it too distracting. The problem with the mode display is that it is showing a text all of the time, making it difficult to spot when it changes to 'anti gravity' without specifically looking at it.

BludzUK, 2017-11-29:
Perhaps it could detect when Anti Grav is active by monitoring the I term. So when anti grav kicks in and boosts the I, the osd warning will see this.

It being distracting isn't an issue imo, you will only have it on your osd while you configure it.

leocb, 2018-06-26:
This has been added on BF3.4 - PR #5686

timman2er, 2018-08-29:
should be closed @BludzUK  .

BludzUK, 2018-08-29:
Didn't realise I had to close it, but thanks for letting me know.

Big big thank you to anyone involved with adding this feature. Puts a smile on my face when I hear people talk about it, and it works great. Also, thanks to the whole dev team for all your hard work.

closed 2018-08-29 by BludzUK