betaflight/betaflight #4688: Feature request changing power of vtx via smart audio using a spare channel

nigelsheffield, 2017-11-28:
Hope this is the right way to bring this up, please advice if it's not.

If it was possible to change the power output of vtx via a spare channel input it could be switched via a logic switch based on rssi levels or arm switch enabling reduced output to save overheating problems when sitting on floor between flying,  it might also help with swamping of other channels.
And of course could simply be changed manually when less power is needed giving more flexibility to the user.
It's just an idea but what do you think?

etracer65, 2017-11-29:
Partial duplicates with:

https://github.com/betaflight/betaflight/issues/3094
https://github.com/betaflight/betaflight/issues/4334

Although adding in an "adjustment" to link to a channel to control VTX power might be useful.  Not everybody runs an OSD and this would enable those without to adjust VTX power on the fly.  Kind of like controlling rate profiles with a an adjustment.

rhys100, 2018-03-30:
X10s has a 6 way knob right in the middle of the controller. pretty sure we all know what its for!
pitmode/25mw/200mw/500mw/800mw/5WATTS!

mikeller, 2018-03-30:
@rhys100: Nonono, the 6 way knob on the X10s is for the 6 rateprofiles that Betaflight 3.3 supports. :stuck_out_tongue_closed_eyes: