betaflight/betaflight #4689: Feature Request: OSD VR combat

riquezjp, 2017-11-29:
Is it conceivable to have a switch set to virtually "shoot" at other quads?
The OSD could be used to display the laser dots, if a HIT is registered then the other quad disarms or has max throttle 50% or some glitch out of the OSD obscuring vision etc.

Obviously the hit detection & linking to the other quads is the issue - but I thought perhaps there is IR sensor or something which could be easily used instead of some special hardware.

Its just an idea. I expect people have thought of it before..
But it seems like an exciting area to explore & develop.

jflyper, 2017-11-29:
IR emitter and receiver would do it.

Ref:
https://github.com/betaflight/betaflight/pull/1931#issuecomment-272614493

bkeinert, 2017-12-04:
Other quads are pretty hard to find in the air. IMHO - to make this actually work (in a way that is interesting) you would want some kind of "radar" in the OSD, showing where other quads are.

But it could be fun and damage models should be rather easy to integrate - making a quad behave bad in the air is just a matter of changing some values.

riquezjp, 2017-12-04:
Finding targets would be very hard I agree. A tail light might make it easier, or a scanner as you say, great idea. 
Depending where & how you 'play' would also make it easier, following a track makes the position of enemies more predictable. 
Aim assist? ha ha ~ it could work, keeping the other quad in view for 3 seconds while you "Lock On"

ServusServorumDei, 2017-12-04:
I have built a small 3D mapping system - in a predefined area <500m^3, it
will actively send location information for all aircraft. IFF, will need to
be implemented somehow to complete a TACAN. I proposed this to an air park
for multirotors: dog fight arena.

Our flight controllers will need more processing power, possibly run F4s or
F7s in parallel.

mikeller, 2017-12-04:
@ServusServorumDei: Just out of interest: What does it use to locate the craft?

ServusServorumDei, 2017-12-04:
I concur. Live and low latency tracking for multiple aircraft is costly:
[financially] hardware, software and development of both.  If you ask any
EE or RFE with experience in radar theory, they will tell you the same.

ServusServorumDei, 2017-12-04:
Multiple (low power) radar installations, four for my experiment to
triangulate (highly reflective objects). All four units had ~180°
horizontal / ~160° vertical Field of view. Picture four of these at the
perimeter of the 'combat theater' pointed at 45° in field and around 90°
vertically so all the installations overlap, and can cover a pre-defined
flight ceiling.

ServusServorumDei, 2017-12-04:
I guess I needed to specify that not only should the aircraft be reflective
(which it should be), but must also have a transponder for IFF.

mikeller, 2017-12-04:
@ServusServorumDei: Wow, that's impressive. Any documentation on this publicly available?

ServusServorumDei, 2017-12-04:
Oh, jeez. I never went public with this after my proposal was declined (the
park was and still is at a startup stage) they didn't have the resources
[and neither did I, :) to finish the IFF system]

mikeller, 2017-12-04:
Hmm, package up what you've got, it will still be inspirational for others.

I can see more applications than 'drone fight club' for this. Considering that most GPS systems seem to turn to custard for high-G applications, being able to graph exact flight paths would probably be of high interest to the racing / PID controller optimisation community...

ledvinap, 2017-12-04:
@mikeller : There is probably reason why Vicon is used in  such scenarios .. even with that price tag

mikeller, 2017-12-04:
@ledvinap: The price tag is something that can easily be changed by the addition of competition for a market segment. ;-)

ServusServorumDei, 2017-12-04:
Vicon IS expensive. I may try this with lidar since my home-made sodar is not as efficient.  @Mikeller, I'll see what I can do - It will be a tight squeeze for time. But seriously, what could be as fun as a virtual dog fight? LOL

mikeller, 2017-12-04:
@ServusServorumDei: This: https://www.youtube.com/watch?v=MZs9DF0m6Xo

ServusServorumDei, 2017-12-04:
@Mikeller Hahahhahahahha!

ServusServorumDei, 2017-12-04:
@riquezjp @jflyper: As I normally do,  I put way too much thought into that with 3D mapping, IFF, and hardware requirements, etc.  Primitive air combat had no such systems (WWI) like biplanes with Rolce Royce engines and all. It could still be done until I finish documentation and rebuilding the mapping system. For example, a miniaturized laser tag system that relays hits to your/the opponent's craft could more than do the trick. The IFF could quite simply be substituted with LED colors for teams (yes,  but remember,  friendly fire isn't... for those of you, like my brother that enjoys Destiny - it's not free for all.....or is it?  *insert Ominous laughter here*

riquezjp, 2017-12-05:
Theres seems several versions possible with various levels of detail & technology. Very cool!

At the most basic: Could the OSD be programmed with a motion detect/tracking feature? 
(that can lock on to bright tail-light LED)
In this case a game could be played to shoot targets for points only, without anything required other than LED. Might not be viable due to fast moving.

The next level, mini-laser tag sounds like it would be really popular & worth the addition of some parts/cost. Its ideal for a few friends or even solo play.

And your mapping tech with IFF sounds like the ultimate tactical area combat! Quad Wars! 😃

mikeller, 2017-12-05:
@riquezjp: The circuitry that is used to generate the OSD for Betaflight onboard OSD is a simple video overlay generator, with no capability to digitise the video. So detection with the OSD circuitry is out of the question.

ServusServorumDei, 2017-12-05:
@riquezjp, what @mikeller said

sevet, 2023-07-11:
I guess this could use the inav radar feature ive requested which uses elrs hardware to communicate location, but this feature needs a bit more accuracy, i gues you could also use it without accuracy just have the craft less than 10 meters from you and in the center of a osd when you shoot to consider a hit, but that feature need to be implemented first, there is actually already a fork of betaflight with formationflight so it shouldnt be too complicated i hope.

closed 2026-09-26 by blckmn