betaflight/betaflight #4690: Random blips in RCCommand throttle in Blackbox

braeden123, 2017-11-29:
Throughout my flight, there are little ~10% up or down throttle movements that are less than .1 seconds in duration.

https://imgur.com/NGcUCKx

Setup:

RS2306 Motors
Frsky XM+ and Taranis X9D
TattooF4s AIO FC and ESCs

## Betaflight diff is below and blackbox log is attached. Look at log 4/4 for the throttle issues.
[BTFL_BLACKBOX_LOG_BSMITH0_20171128_162101.zip](https://github.com/betaflight/betaflight/files/1514265/BTFL_BLACKBOX_LOG_BSMITH0_20171128_162101.zip)

```
# diff

# version
# Betaflight / LUXF4OSD (LUX4) 3.2.2 Nov 11 2017 / 00:34:30 (77782e602) MSP API: 1.36

# name
name BSMITH0

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01

# mixer

# servo

# servo mix

# feature
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1725 2100
aux 1 1 1 1725 2100
aux 2 2 1 1300 1700
aux 3 28 0 1800 2100
aux 4 35 2 1775 2100
aux 5 36 3 1300 1700

# adjrange

# rxrange

# vtx

# rxfail

# master
set baro_hardware = NONE
set rssi_channel = 16
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set align_board_yaw = -90
set yaw_motors_reversed = ON
set small_angle = 180
set osd_cap_alarm = 900
set osd_tim1 = 769
set osd_vbat_pos = 44
set osd_rssi_pos = 2401
set osd_tim_1_pos = 2413
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 2425
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2444
set osd_mah_drawn_pos = 2456
set osd_craft_name_pos = 2092
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 449
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1

# rateprofile
rateprofile 0

set rc_expo = 35
set rc_expo_yaw = 30
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 74

```

DieHertz, 2017-11-29:
Unless your Tx gimbals are faulty, whatever you see in RC command been commanded by your fingers

braeden123, 2017-11-29:
@DieHertz 

It's too fast to be commanded from my radio. Less than a 10th of a second and consistent. Perhaps my gimbals could be dying, anyway to test.

jflyper, 2017-11-29:
You have both UART1 and UART3 enabled for serial RX. Fix this first.

braeden123, 2017-11-29:
@jflyper Okay, good catch thanks. Do you think that might be the cause of the issue?

jflyper, 2017-11-29:
It usually doesn't work at all ...

closed 2017-12-03 by braeden123