betaflight/betaflight #4691: Will not exit turtle mode

LuckyFPV, 2017-11-30:
BF 3.2.2
   I have dual switch arming , when I use turtle mode, flip over and exit turtle, rearm like normal, props are then not spinning in the correct direction . if i wait say 30 seconds after exiting turtle mode before rearming it seems to be ok and the props spin in the  correct direction and I can take off.

mikeller, 2017-11-30:
What firmware version are you running on your ESCs?

LuckyFPV, 2017-11-30:
16.7, timing thing for sure . If I go straight away to rearm , not like lighting fast mind you.  just flipping switches props remain reversed.  if i pause 20 secs after disarming from turtle mode and rearming its ok.

mikeller, 2017-11-30:
Still weird, the reversal commands are sent before arming, so the ESC should be able to process them no matter what.

LuckyFPV, 2017-11-30:
tried a different sequence just now and seems ok
flip over thent come out of turtle mode , then disarm, then immediatly rearm appears to be working .
I may have been flipping over in turtle mode , then disarming and then moving turtle mode switch back to normal off postion .. then rearming and that is were i was experienced the reversed props.

LuckyFPV, 2017-11-30:
perhaps the fact that I was disarming in turtle mode, something was cached as I was still powered up and the slight delay was enought for BF to figure it out ??

closed 2017-11-30 by LuckyFPV