betaflight/betaflight #4698: FLIP AFTER CRASH, BF 3.2.2, reverse motor setup

teshko, 2017-11-30:
Motor direction stays the same. My quad has reverse motors on Quad-X setup (1 - CCW, 2 - CW, 3 - CW, 4 -CCW). I'm using F4 Flame FC with Spedix GS30 blheli_32.

```
# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.2 Nov 11 2017 / 00:39:00 (77782e602) MSP API: 1.36

# name
name Perl

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERIAL_TX 11 A08
resource ADC_RSSI 1 NONE

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ANTI_GRAVITY

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100
aux 1 2 1 1700 2100
aux 2 13 1 1300 1700
aux 3 19 2 1300 1700
aux 4 28 0 1700 2100
aux 5 35 2 1700 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set baro_hardware = NONE
set rssi_channel = 8
set serialrx_provider = SBUS
set blackbox_device = NONE
set min_throttle = 1050
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 179
set deadband = 3
set pid_process_denom = 1
set osd_vbat_pos = 2094
set osd_flymode_pos = 2113
set osd_throttle_pos = 2508
set osd_vtx_channel_pos = 377
set osd_current_pos = 2497
set osd_mah_drawn_pos = 2519
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set vcd_video_system = 1

# profile
profile 0

# rateprofile
rateprofile 0

set rc_rate_yaw = 80
set roll_srate = 75
set pitch_srate = 75
```

mikeller, 2017-11-30:
What version of BLHeli32 are you running on the ESCs?

phillipsj70, 2017-12-01:
I have been experiencing this as well. I use reverse props for motor directions. When in upsides and put in turtle mode my motors don't switch direction.

richard-scott, 2017-12-01:
I believe you need BLHeli32 firmware v32.2 or above for turtle mode to work fully?

teshko, 2017-12-01:
I flashed 32.3 on my ESCs, and it is fine now.

Sorry, I was sure it was BF issue, since I did a quick test in normal setup (not reversed) everything seemed to work fine.

closed 2017-12-01 by martinbudden