betaflight/betaflight #4699: F446: spiTransfer returns before transfer is complete?

mluessi, 2017-11-30:
I came across a strange problem on a F446 target. The `spiBusReadRegister` function would always return 0. After inspecting the bus with a logic analyzer, I saw that the CS line would go high before the last byte was complete. As a tests I inserted busy-waits for the transfer to complete, wich fixed it:

    uint8_t spiBusReadRegister(const busDevice_t *bus, uint8_t reg)
    {
        uint8_t data;
        IOLo(bus->busdev_u.spi.csnPin);
        spiTransferByte(bus->busdev_u.spi.instance, reg | 0x80); // read transaction
        while (spiIsBusBusy(bus->busdev_u.spi.instance)) {};
        spiTransfer(bus->busdev_u.spi.instance, NULL, &data, 1);
        while (spiIsBusBusy(bus->busdev_u.spi.instance)) {};
        IOHi(bus->busdev_u.spi.csnPin);
        return data;
    }

I'm not sure why this is happening or if it is F446 related. Any ideas?

mikeller, 2017-11-30:
Interesting. F446 is not (yet) fully supported in Betaflight. It seems to mostly work, but there are areas (like VCP) that do not properly work, making the existing F446 based flight controllers (like https://www.spektrumrc.com/Products/Default.aspx?ProdID=SPMFCF400) unusable with Betaflight. In this respect, it is likely that what you are experiencing is another F446 quirk, SPI is working correctly on F405 and F411.

But thanks for looking into getting F446 supported, and please consider feeding this back into the project through a pull request.

jflyper, 2017-11-30:
@mluessi If you even have a partially working branch, it's nice if you can publish it. :)

mluessi, 2017-12-01:
@mikeller @jflyper this is for a BrainFPV target that uses chibios to support the graphical OSD. I don't think the problem is due to the chibios though, as it was working normally and then the problem appeared with BF 3.2. I can look into it more.

What are the VCP problems you mentioned? I assume it may be due to the clock configuration. On the F446 one has to use a separate clock (PLLSAI) to generate the 48MHz clock for USB when the main clock is 180MHz. The STM32 libraries seem to up to date, so it should work.

I will try to make a simple target without the chibios stuff to further test / debug this

jflyper, 2017-12-08:
@mluessi 
Thanks for mentioning PLLSAI. We now have a working PR for F446 support, which includes Nucleo F446RE (tested with bare board) and Spektrum F400 (untested) #4723

I'll check if the problem develops.

Btw, what is your target board?