betaflight/betaflight #4700: Adjustable AirMode RPM setting

chime13, 2017-12-01:
Hey Guys, I wanted to know if there is a setting or way to decrease the AirMode motor speed? In my case, I have 2 builds that actually float or increase altitude with AirMode enabled. In cases like these, I would like to lower the RPMs in order to decrease lift.
Apologies if this is the wrong place to post such a question.

Cheredanine, 2017-12-01:
Hi, 
Digital idle if you are using serial protocol, min throttle if using pwm based, however it seems likely you have a problem such as oscillations which is causing air mode to increase the throttle from minimum. You may need to look at that as stock setting don’t make quads drift upwards at zero throttle

chime13, 2017-12-01:
At min Throttle (1000) and air mode enabled, the craft will keep bouncing on the ground until I either give it throttle to lift or disarm... during flight with Airmode disabled, I don't have any runaway/oscillation behavior

Cheredanine, 2017-12-01:
This is completely normal, please read the wiki, the point of air mode in part is its effect in the air, it isn’t designed for on the ground, that would be ground mode.
If you are describing behaviour before you take off at all then once again this is indicative of bad oscillations which air mode is fighting

chime13, 2017-12-01:
I've read the wiki - I do understand "AIRMODE" and it's use. It would just be nice to be able to set the minimum AIRMODE throttle independent from min Throttle - kinda like setting an "idle-up". I don't have oscillations, just a very light build with big props. 
If I disengage AIRMODE then all of my crafts will exhibit the "still present bug" of freaking out during a quick bank or roll that started on BF3.1.7. I am now on BF3.2.2.

Cheredanine, 2017-12-01:
So you understand airbmide does not effect the throttle value? It provides full authority to aspec5 changes independent of throttle value?

You complain your craft climbs at idle in air mode and you want to set a different idle in air mode. As above, air mode doesn’t touch the throttle. Secondly since your idle speed should be as low as possible without prop stall, a seperate idle for air mode would be higher, causing your craft to climb faster at min throttle.

What is happening at the moment is there is a problem on your build causing the quad to change aspect slightly. Air mode, having full authority, corrects by adding power to the dipping motors (it can not take power away from an idling motor), as well as correcting the aspect this generates general thrust/vertical momentum. Above min throttle it doesn’t happen because it can lower thrust from th3 rising motors as well as increasing on the dipping, ‘‘tis keeps the aggregation of the thrust constant.

This is why the infamous air mode bounce landing occurs.
In your case it is occurring outside of landing. This is symptomatic of a problem with the build. You would be better to find and eliminate the problem on your build than try and get the software changed to compensate for it (although as explained above, the change proposed would not fix your problem)

chime13, 2017-12-01:
@Cheredanine "This is why the infamous air mode bounce landing occurs."

Not talking about landings....
Thank you for your time

closed 2017-12-01 by chime13

Cheredanine, 2017-12-01:
Please, not trying to have a go, but read it, I said 
**“In your case it is outside of landing. This is symptomatic of...”**

chime13, 2017-12-01:
Your point is understood - thank you.

reicek, 2018-12-23:
It seems you need to reduce the throttle for your air mode, try 10% and see if thats enough to keep them spinning without fly outs.

From the CLI type:

`set airmode_start_throttle_percent = 20`
`save`