betaflight/betaflight #4705: Feature Request - Lost and found information stored on FC

NitroNAF, 2017-12-01:
While it won't solve the issue of your multicopter being lost it does solve the issue of being too lazy to create a contact label to stick on your craft. I believe it would be worthwhile to have a small string of information stored on the Flight controller with at least a phone number so that if a fellow pilot finds your craft they can plug it in and find its owner.

Abancalari, 2017-12-01:
It's been suggested before. Got told there wasn't enough space for it. Then they added the "craft name" value which is probably enough for a phone number. Though the person finding it would have to know to plug it in to betaflight.

claytonbakerjr, 2017-12-01:
not only that but if you run the OSD and type the phone number in the craft name area...you will see it in the OSD. This likely would have to be an addon thing in a separate text field

fiveangle, 2017-12-01:
Label makers work pretty well :)

Only value I can see is if the FC is sold, and then the buyer would see it on power up.  But at that point it's already a legal mess since the buyer paid for the parts.

I have my FCC license callsign set for _CRAFTNAME_ displayed in all my quads, which allows lookup and direct snail mail contact via the [FCC Current License by Callsign Search](http://fjallfoss.fcc.gov/General_Menu_Reports/callsign.cfm) facility, which is the easiest method to meet the 10 min call sign broadcast period requirement.

Doesn't everyone ? =D

mikeller, 2017-12-01:
Just to make a suggestion (I totally agree with @fiveangle that a label strip should be the first line of defence. If you are flying a craft that uses an SD card (either for logging or in a camera), add a text file to the sd card containing your address - even if the finder has no idea about RC flying, they will probably be able to recognise an SD card, and pull it out and stick it into their PC.

Cocobeasdad, 2017-12-25:
I use my phone number as my craft name.  If somebody is smart enough to open Betaflight they'll come across the info.