betaflight/betaflight #4708: Feature Request: Preference reminders.

LlamaB, 2017-12-02:
Ok so I know therte has to be other people out there that have the same problem and I think i have a solutuion.

The problem example : 
The user likes to set up certain things such as air mode, small angle, reverse propellers or other settings. But there can often be a lot to remember, and the worst time to realize you forgot to setup something how you like is after it's too late.

The solution I propose: 
Add a way for the user to program beta-flight itself with the usual settings the user normally likes and uses. When an FC is connected, check to see if those pre-determined settings are how the board is set or not. If the settings are not the user selected normal, simply alert the user and remind them the setting is not their normal. The the user can fix it and be less likely to overlook one little thing.

Hope this make sense.

mikeller, 2017-12-02:
@LlamaB: This is a feature request for the Betaflight configurator, not the firmware. As such, please close it here and open it here: https://github.com/betaflight/betaflight-configurator/issues

LlamaB, 2017-12-02:
Ok, thank you I will.

closed 2017-12-02 by LlamaB