betaflight/betaflight #4710: SPRACINGF3EVO Frsky Sbus NOT Working

sjb0095, 2017-12-02:
Warlark 80 - upgrading betaflight above 3.0.1 and the receiver stops working.

etracer65, 2017-12-02:
I have multiply SPRACINGF3EVO targets with SBUS and they're all working fine.  Verify your settings on the ports tab and that the correct UART is selected for your serial receiver.  Also check that serial receiver and SBUS is selected on the configuration tab.

Hopefully when you "upgraded" you didn't try to use a dump from a pre-3.2 version as that won't work.  You need to set up the configuration from scratch.

sjb0095, 2017-12-02:
Hi, Thanks for replying,

the ports are correct: I've already tried wiping the board and starting fresh - but it still doesn't work!

dump
# dump
# Betaflight / SPRACINGF3EVO (SPEV) 3.2.2 Nov 11 2017 / 00:44:21 (77782e602) MSP API: 1.36
name Warlark
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A15
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource LED 1 B08
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource TRANSPONDER 1 A08
resource SPI_SCK 1 B03
[... 488 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

etracer65, 2017-12-02:
Try `set sbus_inversion = OFF` as a test.  Maybe you have a non-standard uninverted SBUS output from your receiver.

DieHertz, 2017-12-02:
https://www.rcgroups.com/forums/showpost.php?p=38455867&postcount=130

sjb0095, 2017-12-02:
set sbus_inversion = OFF: didn't work  :(

interestingly if I change it from SBUS to spectrum Betaflight receiver tab responds and seem to be getting input - SBUS is blank.

sjb0095, 2017-12-02:
@DieHertz  - Thanks This custom firmware worked! will this ""Odd parity option" be built info future BF updates? or is there a work around for 3.2.2?

DieHertz, 2017-12-02:
I'll build a custom binary for you when I get home.
Full fix is coming to 3.3 as it's a bit more involved than initially planned

sjb0095, 2017-12-04:
@DieHertz : Hey Andrew, did  you manage to create a new binary?

DieHertz, 2017-12-05:
@sjb0095 sorry, I got lost completely :-(
Go into CLI and `set serialrx_parity = odd`
[betaflight_3.3.0_SPRACINGF3EVO.hex.zip](https://github.com/betaflight/betaflight/files/1530732/betaflight_3.3.0_SPRACINGF3EVO.hex.zip)

DieHertz, 2017-12-05:
@sjb0095 Sorry, I got lost completely :-(
Just flash it, it uses odd parity for SBUS by default.
[betaflight_3.2.2_SPRACINGF3EVO.hex.zip](https://github.com/betaflight/betaflight/files/1530777/betaflight_3.2.2_SPRACINGF3EVO.hex.zip)

sjb0095, 2017-12-05:
Tested and works perfectly - Thanks for your help!

DieHertz, 2017-12-05:
Glad to be of help :+1:

sjb0095, 2018-08-16:
@DieHertz  Hey DieHertz - this still seems to be a problem in 3.5 - any chance you can upload a 3.5 version with SBUS FIX?