betaflight/betaflight #4711: hang during `diff` on F3 OMNIBUS

fiveangle, 2017-12-02:
1.) Load 756795721c6d93a64752df5abdfb795c531c8def
2.) Restore configuration
3.) Execute `diff` in cli
4.) Observe hang partially through the output
5.) Observe "always blink" LED strip halt activity
6.) Observe no response until board is reset

Tested on OMNIBUS targets: mini F3 clone (Fortinet v1.0) and full-size AR F3 Pro clone

diff output does not halt at the same command on both boards tested.

Sample from mini F3 OMNIBUS clone (from ARFun95):

```
# diff
# Betaflight / OMNIBUS (OMNI) 3.3.0 Dec  1 2017 / 10:49:49 (dd3a40f) MSP API: 1.37
name AI6DH

feature -RX_PPM
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature MOTOR_STOP
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
beeper -ARMING_GPS_FIX
beeper -MULTI_BEEPS
beeper -ON_USB
map TAER1234
serial 0 1 115200 57600 0 115200
serial 2 68 115200 57600 0 115200
led 0 0,0:W:ABIW:2
led 1 15,0:E:ABIW:2
aux 0 0 0 950 1200
aux 1 1 1 1600 2050
aux 2 13 1 1800 2100
aux 3 15 2 1800 2100
aux 4 19 2 1300 2100
aux 5 27 3 1700 2100
aux 6 28 1 900 1400
aux 7 30 2 1300 1700
aux 8 35 3 1300 1700
rxfail 3 h
rxfail 6 s 1500
set align_gyro = CW270FLIP
set moron_threshold = 32
set align_acc = CW270FLIP
set baro_hardware = NONE
set max_check = 2000
set fpv_mix_degrees = 15
set max_aux_channels = 4
set serialrx_provider = SBUS
set blackbox_device = SERIAL
set min_throttle = 1100
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 400
set failsafe_delay = 5
set failsafe_procedure = AUTO-LAND
set bat_capacity = 500
set vbat_max_cell_voltage = 42
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set current_meter = VIRTUAL
set ibatv_scale = 148
set ibatv_offset = 39
set yaw_motors_reversed = ON
set small_angle = 180
set fixedwing_althold_reversed = 
```

mikeller, 2017-12-02:
Can't reproduce, works fine for me.

A couple of hints:
- use `diff all` and not `diff` to save / restore the config. Restoring a `diff` over a config that has not been reset will not work. `diff all` contains commands to reset the config first;
- if using `diff`, do a flash with 'Full Chip erase', or else a 'Reset Settings' before restoring.

fiveangle, 2017-12-02:
Report is that diff fails on 3.3.0.  diff all is used to backup.

Did another test without resotoration and bombed too:

https://www.screencast.com/t/HXJ1y8UaceEo

To save from having to watch, it bombs at:

`set fixedwing_althold_reversed = 
`
on this FC, but should have been no diffs, so perhaps that's a clue ?  Will try the other board again and see if it bombs at same place.

fiveangle, 2017-12-02:
*drumroll* identical issue with all defaults:  https://www.screencast.com/t/IO4f8mFL

Took a quick look and didn't see anything obvious, although #4619 may be related if `fixedwing_althold_reversed `expects bits for ALT_HOLD to be present 🤷‍♂️

mikeller, 2017-12-02:
Got it!

Your original bug report stated the changeset where this is happening to be 7567957. This is not the case, 7567957 is just fine. Your videos show dd3a40f to be the problematic changeset, and this _is_ correct.

Seems to be non-matching defines for `fixedwing_althold_reversed`. Can you try this to see if it fixes it for you:

[betaflight_3.3.0_OMNIBUS.zip](https://github.com/betaflight/betaflight/files/1524352/betaflight_3.3.0_OMNIBUS.zip)

fiveangle, 2017-12-02:
You're right.  And my original diff output shows the same `dd3a40f` changeset.   Copied hash from wrong commit.  Sorry for the confusion.

Your build above resolves 👍

mikeller, 2017-12-02:
I didn't even look at the changeset in the diff, since you explicitly stated what changeset it applied to. :-P

closed 2017-12-03 by fiveangle