betaflight/betaflight #4712: GPS lock disables RX_SET beeper and dshot beacon.

Harpalyke, 2017-12-02:
Encountering an issue on 3.2.x maintenance branch where once my quadcopters GPS sensor acquires a lock the RX_SET beeper wont make any sound. It works fine before the GPS lock is acquired and then once its locked both the beeper and the dshot beacon tones are disabled.

I came accross issue #4123 which suggests to enable `beeper MULTI_BEEPS` via the cli. However enabling this flag only had the effect of enabling the beeper to beep the number of satellites that are connected rather than enabling the RX_SET beeper and dshot beacon again.

I’ve looked over the code and it seems that `IS_RC_MODE_ACTIVE(BOXBEEPERON)` is reporting false within the beeperUpdate function. Cant seem to track down where this is being unset by the MULTI_BEEPS flag though.

mikeller, 2017-12-02:
@Harpalyke: Yes, this is a drawback of how the beeps for RX_SET and GPS lock are mixed in 3.2. In 3.3, this was improved by the introduction of a new `BEEP GPS SATELLITE COUNT` mode that is independent of the RX_SET mode. But since this is a new feature, it will not be backported into 3.2.

Harpalyke, 2017-12-03:
@mikeller: Ah OK, sorry. I see the PR for that there now. Thanks!

closed 2017-12-03 by Harpalyke