betaflight/betaflight #4714: Documentation update needed for Linux install of Betaflight

caj411, 2017-12-03:
Current versions of Linux come with all the necessary drivers installed. The issue for most users attempting to use betaflight in Linux is the inability for betaflight to connect to the serial port.  In most instances it is an easy fix, nothing more than a file permission issue.

In my opinion, it should be at the very top of the Betaflight Linux install section, highlighted in flashing text, to prevent people from spending countless hours going down rabbit holes, like myself and many others.

credit goes to Chris GR on youtube..   https://youtu.be/JpeXEb9pLZE

THE SOLUTION..
sudo chmod +666 /dev/<your_port> 
example: sudo chmod +666 /dev/ttyUSB0

mikeller, 2017-12-03:
https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#platform-specific-linux

Also, note that giving full access to everybody to whatever tty you are using (which is dynamically allocated anyway) is an insecure hack.

closed 2017-12-03 by mikeller

caj411, 2017-12-03:
Most of us are using our personal machines at home behind routers and
firewalls where some hacks are acceptable, I understand your position, but
for myself and many others it works just fine to get betaflight working.

Could you at least clarify what is there, please. As it is currently
written, it is a bit technical for the average user, and in some places
ambiguous and incomplete.  The install section doesn't mention the fact
that the drivers are already installed on most current Linux installations,
that in itself would prevent many people from trying to install drivers
that are unnecessary and save a lot of wasted time and effort.

Per your page,

https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#platform-specific-linux

The current instructions say this...

*Platform Specific: Linux*

*Linux requires udev rules to allow write access to USB devices for users.
The command bellow will create a template rule for you.*

*(echo '# DFU (Internal bootloader for STM32 MCUs)'*
* echo 'ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483",
ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev"') | sudo tee
/etc/udev/rules.d/45-stdfu-permissions.rules > /dev/null*

*Now you need to find the real product id of your FC. Type in the command
bellow and plug your FC in and out. It should print a line with the product
id out. *    *udevadm monitor --environment --udev | grep ID_MODEL_ID*

*that is clear, no problem there.
*my example return:  ID_MODEL_ID=ea60

*Now update the entry in "/etc/udev/rules.d/45-stdfu-permissions.rules"
accordingly. You can add more than one rule in the file. The default
product id is the FC in bootloader mode.*

**this is "muddy".
Can you define what a rule is?
Is the text *ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483",
ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev" *a single rule?
There isn't any more information regarding additional rules. Why would I
need another rule?

Do we have to put it in bootloader mode to get the FC product ID?
Does the product ID change depending on if its in normal vs. bootloader
mode?
Why do we care, why is that last sentence important to know, I'm assuming
that statement was put there for a reason?

cat /etc/udev/rules.d/45-stdfu-permissions.rules
*# DFU (Internal bootloader for STM32 MCUs)*
*#ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483",
ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev"*

*my example return:  ID_MODEL_ID=ea60

*Where in the 45-stdu-permissions.rules file does it say ID_MODEL_ID? The
ID_MODEL_ID is not a listed attribute. Where do we put the ID_MODEL_ID? Do
we add the attribute to the list of existing attributes? If so, what
format? ID_MODEL_ID=="ea60"?

Sorry for all the questions, but just trying to clarify.

mikeller, 2017-12-03:
Fair enough. The text seems to be fairly straightforward to me. But I appreciate that it might not be appropriate for all levels.

Can you please propose clarifications where you think they are needed, so that we can add them?

caj411, 2017-12-05:
> Let me give it a shot, it I think it could be made clearer for the average
> joe.  I'll get back to you.

Micheal,

Attached is my attempt to rewrite the Linux installation instrucitons,
where a non-technical user can follow along with practically no Linux
knowledge. I have written it using LibreOffice, so I hope that format
(.ODT) will work for you.

Where you see yellow highlights, these are things I would add, BUT I am not
sure if those statements are correct because of unanswered questions that I
have myself.  If you can provide me with those answers where I have
questions, I can finish the document in a complete form.

Where you see Red text is simply comments by myself where things are
incomplete and lacking details that I have no idea what to document.
Again, if you will answer those questions, I can produce a complete
document, or you can fill in those deficiencies yourself.  I am trying to
leave out all assumptions and make the installation effortless for
everyone. I am a firm believer that you should provide instructions for the
least common denominator, and in today's world we have potatoes using
computers, so that bar is very low.

I have not done anything to the troubleshooting section, except added my
insecure hack as a "if all else fails" option with warning.  The
troubleshooting is beyond the basic installation,  but I would suggest
looking into re-writing that section following my format of no
assumptions,monkey see, monkey do. Things written specific to SystemD
should be noted, because some people may have distributions that are not
using SystemD, it is not yet included across in all distributions.

The biggest assumption people make is that ALL Linux users are highly
technical, that is simply not true. They should be afforded the same level
of assistance that Windows users are afforded, the assumed computer
knowledge of a potato.

Thanks for listening to my concerns and allowing me the chance to help
improve what I see as documentation deficiency.  I have seen a plethora of
posts from people unable to get betaflight to work with Linux, hence my
suggesting that the simple chmod 666 hack will work if all else fails.  I
see no reason to withhold any pertinent information as long as it is
explained and documented properly.  The goal is to expand the userbase of
betaflight and ease of use is the best way.

mikeller, 2017-12-06:
It looks like attaching did not quite work - you might have to zip it to get GitHub to accept it.

caj411, 2017-12-06:
Here you go.  See if this works.

[betaflight_linux_install.zip](https://github.com/betaflight/betaflight/files/1533877/betaflight_linux_install.zip)

caj411, 2017-12-20:
I never received a response with regard to the above documentation.  I would like the questions answered if possible, even if the information is not used to update the website instructions.  Thanks.

mikeller, 2017-12-20:
Sorry @caj411, this has fallen through the cracks. I will look into incorporating your suggestions in the next few days.

caj411, 2017-12-20:
Thanks, as long as the average person can follow the directions and get the
install to work for Linux we're golden.  It doesn't matter how hard the
programmers work to improve the product if the end users have problems with
the documentation, particularly the install.

mikeller, 2017-12-21:
> It doesn't matter how hard the programmers work to improve the product if the end users have problems with the documentation, particularly the install.

True. But since it's the same developers that also maintain the documentation, there's only so much time they can spend on each of these, since it's all done in their free time.

caj411, 2017-12-21:
Seriously?  Wow, that's a full plate.  I'll help where I can if needed.  I
spent many years translating between code talkers and management.

mikeller, 2017-12-21:
Yes, Betaflight is not a commercial project. ;-)

Your offer is appreciated, and probably the best way for you to get to work is to have a look trough the wiki (https://github.com/betaflight/betaflight/wiki), and see if you can find other areas where documentation can be updated / improved, or where you think it is missing something. The wiki is editable by everybody with a GitHub account.

You can also join the Betaflight Slack (http://betaflight.ch/), and help other users with their configuration issues in there.

martinbudden, 2017-12-21:
@caj411 , any help in updating and improving the documentation is most welcome. Thanks.

mikeller, 2017-12-22:
@caj411: Done: https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#platform-specific-linux

h89434, 2018-01-15:
# xubuntu xenial (16.04.2 LTS)
quad@quad-m:/dev$ sudo gpasswd -a quad dialout
quad@quad-m:/dev$ ll /dev/ttyA*
crw-rw---- 1 root dialout 166, 0 jan.  15 19:58 ttyACM0
quad@quad-m:/dev$ #sudo chgrp dialout ttyACM0 
#Example matekf405
quad@quad-m:/dev$ lsusb -t
quad@quad-m:/dev$ sudo nano /etc/udev/rules.d/45-stdfu-permissions.rules 
quad@quad-m:/dev$ cat /etc/udev/rules.d/45-stdfu-permissions.rules 

# DFU (Internal bootloader for STM32 MCUs)
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0664", GROUP="dialout"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE="0664", GROUP="dialout"

Now in chrome:betafight-configurator, use /dev/ttyACM0 115200 , Connect

caj411, 2018-01-17:
I followed the process step-by-step and it worked perfectly. Thank you for updating the documentation!  

 do have one more additional question.  The very last sentence in section 2 of the Linux install states "Now you will need to get the product ID for your specific FC."  Why is the FC product ID needed? It would appear to me this sentence should be deleted from the document.

mikeller, 2018-01-17:
@caj411: Thanks, fixed.