betaflight/betaflight #4719: Frsky F.port testing.

chaotixtech, 2017-12-05:
Having a spare xsr-m and a Pikoblx F3 I decided I would experiment with this new protocol.
I have flashed the f.port firmware to the xsr-m and have the latest nightly flashed to the Pikoblx, connected the xsr-m via the smartport signal wire, to uart2 on the Pikoblx which is the one used for smartport so should be suitable.
I have configurator 10.0.0 rc4 and set up uart2 as serial rx port and F.port as the serialrx provider, all as per the instructions in the wiki. 
I have a successful bind and correct responses in the receiver tab so the rx side seems fine.
The problem is I am unable to get any telemetry back from the fc to Taranis, the usual streams from the rx are there but attempts to discover new sensors always fails.
There seems to be very little discussion of F.port online so far so my problem solving is limited.

Pikoblx F3 BF3.3.0
Frsky xsr-m F.port firmware
Taranis xd9+ OpenTx 2.2.0

```# diff
# Betaflight / FF_PIKOBLX (PIKO) 3.3.0 Dec  5 2017 / 06:12:58 (49ba30174) MSP API: 1.37
resource MOTOR 1 A07
resource PPM 1 NONE

serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
set serialrx_provider = FPORT
profile 0

rateprofile 0

#```

mikeller, 2017-12-05:
@chaotixtech: https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol#installation

chaotixtech, 2017-12-05:
@mikeller I followed those instructions as they are basically all there is. I've been playing about with variations of tlm_inverted with no success.

mikeller, 2017-12-05:
@chaotixtech: Please restore your initial post to its original content. I am not going to discuss this further if you keep changing the content of your statement.

chaotixtech, 2017-12-05:
@mikeller sorry, all I changed was the "diff" to reflect an actual flyable set up, adding dshot600, TAER1234, min_check = 1010, aux1 for arming,  tlm_halfduplex = OFF.
I will add as a new post in future, would you like me to set defaults and attempt to recreate the original "diff"?

mikeller, 2017-12-05:
For next time, rather add a new post than changing the essence of the old one - it's hard to diagnose something if it keeps changing.

- why are you playing with `tlm_inverted`? This has no bearing whatsoever on FPort.
- if you read the instructions, why aren't you setting `serialrx_halfduplex = on`, which is what's needed to make this work on F3?

chaotixtech, 2017-12-05:
As telemetry is the part not working with my current settings, I thought tlm_inverted had something to do with telemetry so was trying it.
When I ```set serialrx_halfduplex = ON``` I lose receiver connection.
I am using an unaltered xsr-m, no inversion hack and the instructions are not entirely clear regarding suitability. I presume the reason for using f.port would be to not require any inversion hacks and allow use across F3 F4 and F7.

chaotixtech, 2017-12-05:
@mikeller I must report complete stupidity on my part, misread the silkscreen on the fc. 
This is now working as I'd hoped, Thanks.
I will test over the next week to ensure there are no obvious problems.

mikeller, 2017-12-05:
@chaotixtech: Have you connected a battery? There's been a change for VFAS to be only reported when a battery is connected - I think you've just found a problem with that (i.e. VFAS not being discovered in OpenTx if no battery is connected).

Also, thanks for testing.

chaotixtech, 2017-12-05:
This is with a 3s connected, I've deleted and rediscovered but it is still missing.

mikeller, 2017-12-05:
Do you have a SmartPort setup that you can test with the same build? The telemetry code is largely the same, so I suspect that this is not specific to FPort.

egvneto, 2017-12-05:
I'm building my first quad from scratch using the fc HGLR F4 V5PRO (the other ones were RTF). My doubt is, to test this feature, I must not connect SBUS pin of the XSR receiver to the SBUS in my FC, instead I have to connect the S.Port pin of the receiver to the SBUS pin of my FC (in my FC this is UART1)? 
I can't do that tiny soldering hack with my skills.

chaotixtech, 2017-12-05:
@mikeller This fc seems to be faulty as there is no voltage showing in betaflight either, I'll test with another stock xsr-m and standard s.port just incase.

mikeller, 2017-12-05:
@egvneto: No you have to connect the SmartPort pin on your RX to the 'SmartPort' pin on your flight controller (exactly like you'd be doing when using SmartPort.)

egvneto, 2017-12-05:
Ok. @mikeller that would be the TX3 or TX6, I just don't know if these ports are "bidirectional inverter" as the tutorial says and the FC manual is really short in informations. 
I'll try it anyway. Thanks for the tip!

mikeller, 2017-12-05:
@egvneto: You'll have to ask the manufacturer. But chances are that none of them is. It might be a better approach to directly go for the 'uninverted SmartPort' hack (https://oscarliang.com/uninverted-sbus-smart-port-frsky-receivers/).

chaotixtech, 2017-12-05:
@mikeller, I replaced with a Piko that has working voltage in betaflight and VFAS now shows up along with all other sensors. As far as I'm concerned this issue is closed. Thanks.

closed 2017-12-05 by chaotixtech

egvneto, 2017-12-05:
@mikeller I went to hack... it worked... shaking fingers and all that stuff... happy with my setup with 100% telemetry working. Thanks.

Snake675, 2018-01-05:
@mikeller: Thanks a lot for this feature. Thats the future! I used the spare UART for ESC telemetry, nice! Telemetry and lua scripts work fine.
Setup: DYS F4 Pro AIO, FrSky XSR, Taranis x9d+