betaflight/betaflight #4720: Dshot broken 3.2.2 omnibusf4sd

kitesteez, 2017-12-06:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6.) Remove this Text :).

kitesteez, 2017-12-06:
Had to downgrade to 3.2.1
Dshot not working for omnibus f4sd

emski0, 2017-12-06:
im using dshot 1200 on omnibus f4sd 3.2.2, no problems for me

kitesteez, 2017-12-06:
Could be my combo of fc and esc then? I tried downgrading to 16.67 in
blheli as well. Multishot worked, but dshot didn't until I went back to
3.2.1

claytonbakerjr, 2017-12-06:
Are you using the HobbyWing XRotor FC/ESC combo?
The 4in1 ESC only supports DSHOT 600

kitesteez, 2017-12-06:
No I'm using hglrc 4-1 28a esc

jflyper, 2017-12-07:
@kitesteez 
I've tested 3.2.2 with several OMNIBUSF4SD target boards including Omnibus F4 V5 (with Wraith32) and Asgard V2, and they all worked fine with Dshot (F4V5 upto 1200, Asgard up to 600).

I will close this now as your issue is likely to be hardware or configuration related.

If you still believe it is a software issue, please open a new issue ticket.
When you do so, _**please read and follow the instructions**_ given in the text field when you open the ticket.

closed 2017-12-07 by jflyper