betaflight/betaflight #4724: Futaba T16SZ SBUS AUX channel dosn't work

thejackp8, 2017-12-07:
Hi everyone,
I have configurate my new quad, I have alredy flight, but i can't use AUX channel exept 4 for set modes
When I use with the R7008SB sbus2 I can set up 4 aux ch more but not 1to4.
Is not a tx problem I have setting the switch in mixer. 
-max channel in cli are set to 14 (max)
-map AETR1234

-Betaflight 3.2.2
-FC F4 V5pro
-TX Futaba t16sz
-RX cooltec  RSFSB
-RX Futaba R7008SB

mikeller, 2017-12-08:
@thejackp8: GitHub is for questions and discussion around the **development** of the firmware and tools. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?query=betaflight%20cleanflight%20configuration%20use).

closed 2017-12-08 by mikeller

jflyper, 2017-12-08:
@thejackp8 
FYI, I confirmed T14SG + R7008SB using S.BUS2 port, bound in FASSTest 14CH Single mode has no problem utilizing entire channel range.