betaflight/betaflight #4725: [Feature Request] 3D Horizon Mode

Aileron8, 2017-12-08:
It would be nice to have a 3D mode that was initiated via the acc from Horizon mode.

jflyper, 2017-12-08:
What do you expect if sticks are released when an axis is vertical?

Aileron8, 2017-12-08:
I’m not sure I understand your question? If there was a 3D horizon mode I was thinking Id probably reduce the horizon strength to zero while leaving the default transition to 100%. So effectively the quad would still act like acro, but would use the acc to auto-trigger the reversal of motor direction when the small_angle threshold was exceeded, e.g., when inverted. Essentially infinite inverted hangtimes after a 180deg roll or flip.

Aileron8, 2017-12-08:
If Throttle was released when inverted then Airmode would still be active, meaning it should still fall towards the ground, but at a reduced rate. Then if the pilot adds Thrust it should hover. Similar to 3D mode now though, if the pilot used a switch to disable 3D horizon then they could do a roll or flip like normal, but this would enable 3D maneuvers without using a switch during the act of pitching/rolling.

fftunes, 2017-12-09:
During motor transition the quad is basically falling dead for a short moment. If you make the acc have control over motor transition, you have no control for when it happens. It could happen at the worst times... 
I mean, for example if you would want to do a full normal roll, motors would reverse twice while doing so, and during those reversals there's no control or correcting error. Combined with some twitchy stick commands and zero protection against unsynced reversals things could get out of hand quickly.

ethomas997, 2017-12-09:
If you set  `horizon_tilt_expert_mode` to ON then the self-leveling correction will be basically turned off while the craft is inverted, which may be what you want for 3D flying.  See here:  https://github.com/cleanflight/cleanflight/blob/master/docs/PID%20tuning.md#horizon-mode-commands

Aileron8, 2017-12-11:
Yes, enabling horizon_tilt_expert_mode sounds like the optimal way to go about it. I don’t think it would be twitchy using the acc if used as a switch mode. I’ve played around with 3D mode myself and find it difficult and twitchy to trigger using a switch as it stands today though, and using the central throttle is not a natural experience at all imo. While I’m sure with enough practice the latter switch enabling can be mastered I think using the acc would make 3D mode more user friendly in the meantime. I’d use a switch to disable 3D horizon mode before executing regular flips and rolls, then letting the acc trigger motor reversal on its own when 3DHM is enabled.

fftunes, 2017-12-11:
I assume you want the acc to trigger reversal at 90 degrees. What if the quad twitches enough on reversal to get back to 90 degrees.

All i'm saying is, the acc-triggered reversal would need some extra safety checks at least.

Aileron8, 2017-12-11:
Good point. Playing that scenario out, if during a fast roll the acc triggered a twitchy flip severe enough that it re-initiated forward motor direction than it is possible that an oscillation could develop where a subsequent twitchy flip might then reverse again, and the cycle continuing until eventually the reversal sticks due to the RCcmd in roll decaying the oscillation to zero and the acc detecting a steady inversion.

Perhaps there could be some type of interrupt driven debouncing method applied to minimize the odds for the latter oscillation to develop? Or even a small delay at the inversion point might be enough as well.

Aileron8, 2017-12-31:
After giving this more thought I believe having a 3D Horizon mode may be the best way to use 3D mode in terms of stability and smoothness, only using the acc to trigger the timing of PIDs being shutoff and motors reversed; nothing more. 

I’ve been doing a lot of modeling and am convinced much of the twitchiness around 90deg is due to allowing the quad to reach the max pitch angle before off-axis thrust is reduced; props reversed before tilt reaches 90deg. The max pitch angle = arccos(weight/thrust) and can be anywhere between 60deg to 90deg for racing quads depending on where the Throttle is set, the total weight of the quad, and the motor/prop configuration being used. Pitching or rolling past the max pitch/roll angle is automatically corrected and compensated for by the FC, but leads to large spikes in the traces and is summed over time, but cannot be tuned out. Killing the PIDs before reaching the max pitch/roll angle minimizes the spikes as does adding a delay when reversing the props at the 90deg point, but I believe using the accelerometer along with two user data points, weight and approx thrust, along with throttle % the PIDs could be chopped and prop-reversed to coincide just before the max pitch/roll is reached for a smoother transition, customized for each quad. Quads with high T/W ratios would receive the least amount of delay at the highest throttle % while quads with low T/W ratios could receive the largest delay, which would noticeably reduce overall throttle capability, but might be able to minimize the spikes plaguing 3D transitions for larger/heavier quads especially.

Aileron8, 2017-12-31:
[https://www.rcgroups.com/forums/showthread.php?2645953-Mini-tubular-and-ultralight-LOS-Acro-FPV-Quads/page220](url)

closed 2026-09-26 by blckmn