betaflight/betaflight #4727: [Feature Request] FPort for R-XSR

dcanham, 2017-12-08:
Any chance fport for r-xsr will be available? I noticed that it does run a different MCU (than xsr. x4r, or xsr-m) so that will probably add some more work.

DieHertz, 2017-12-08:
Ask FrSky, not BetaFlight

dcanham, 2017-12-08:
Yeah I noticed in the wiki it was a collaboration of the two. I'll be sure to shoot them an email as well.

mikeller, 2017-12-10:
Thanks, @dcanham. It has been brought up in discussions with FrSky that getting firmware that supports FPort for more receivers would be desirable. Getting feedback from users will make a good point that there is demand for this.

xeniter, 2017-12-10:
perfect, can we anyhow help?
would also like to have fport for the r-xsr

will start now testing it with xsr

mikeller, 2017-12-10:
@xeniter: Yes, contact FrSky and ask for it. :-)

closed 2017-12-10 by mikeller