betaflight/betaflight #4728: BB Logs not readable on AnyFC F7 when set to 1.5kHz (1.0kHz is OK)

richard-scott, 2017-12-09:
BB Logs not readable on AnyFC F7 when set to 1.5kHz (1.0kHz is OK).
Sample logs below:

[1.0kHz.BFL.log](https://github.com/betaflight/betaflight/files/1544664/1.0kHz.BFL.log)
[1.5kHz.BFL.log](https://github.com/betaflight/betaflight/files/1544665/1.5kHz.BFL.log)

jflyper, 2017-12-09:
You are not giving us enough information.

mikeller, 2017-12-10:
I think there are other tickets open about 1.5 kHz logs not being readable (I can't find any at the moment). This does not seem to be flight controller specific.

richard-scott, 2017-12-10:
@jflyper - ok, I'll explain more. I recorded some logs at 1.5kHz and I can't load them in the BB Log Viewer. I have tried both the one from the Chrome Store, and I have also checked out the latest repo from GitHub and used that too. Neither versions read the file. If I lower the logging to 1.0kHz, then I can record and read the logs created at that speed.

@mikeller - The only report I can find is this one #3917 they suggest removing and re-adding the BB Log Viewer which I have sort of done. I've not removed it, but I've added the latest direct from GitHub.

Hence why I loaded the logs onto the OP and hoped that someone else may look and say "oh shit, I can't read the 1.5kHz one either"... or "nah, your talking crap I can read them both".

If you don't feel it is a BF issue then I'll open a ticket with the BB Log Viewer repo.

closed 2017-12-10 by richard-scott

jflyper, 2017-12-11:
@richard-scott Which firmware are you running???

richard-scott, 2017-12-11:
v3.2.2