betaflight/betaflight #4731: FPV Camera Angle not working

dgrigg123, 2017-12-10:
I have changed the FPV Camera Angle from 20 degree to 40 degree but do not see any change in the center of rotation when I do a high speed roll.
Not sure if I am crazy, but I just do not see that the center point of rotation moves up/down within the FPV FOV as I change this variable.  It seems to remain at the lower edge of the FOV.  I have a camera tilt of about 32 degree, but changing this variable does nothing to change the rotation location in the FOV.

I am on BF 3.2.2 on a CLRACING F4.
Below is my latest DIF.

# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.2 Nov 11 2017 / 00:24:23 (77782e602) MSP API: 1.36

# name

# resources
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 7 1700 2100
aux 1 26 1 1700 2100
aux 2 27 5 1700 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_trim_pitch = 24
set acc_trim_roll = 24
set rssi_channel = 14
set fpv_mix_degrees = 40
set serialrx_provider = IBUS
set dshot_idle_value = 475
set motor_pwm_protocol = DSHOT1200
set failsafe_kill_switch = ON
set align_board_roll = 180
set align_board_pitch = 180
set pid_process_denom = 1
set osd_vbat_pos = 2071
set osd_rssi_pos = 2113
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2081
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2166
set osd_mah_drawn_pos = 2134
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 2342
set osd_pid_pitch_pos = 2374
set osd_pid_yaw_pos = 2406
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2103
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2250
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_tim_2 = OFF
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 4000
set dterm_setpoint_weight = 100
set p_pitch = 80
set p_roll = 75
set p_yaw = 90
set i_yaw = 50

# rateprofile
rateprofile 0

set rc_rate = 155
set rc_rate_yaw = 155
set rc_expo = 18
set rc_expo_yaw = 10
set roll_srate = 82
set pitch_srate = 80
set yaw_srate = 75

#

etracer65, 2017-12-10:
You also need to enable a flight mode for "FPV ANGLE MIX" on the modes tab.  The function is only active while the mode is selected. This way you can switch it on and off in flight.  If you want it on all the time, simply set the slider range from min to max.

dgrigg123, 2017-12-10:
Thanks for that.  Now I feel stupid, but I don't remember pre-3.2 requiring a mode to be set to do angle mix as it just needed to be set in CLI.   Am I wrong that this has changed a bit?

etracer65, 2017-12-10:
I believe it used to be a feature and was changed to a mode to allow it to be enabled/disabled in flight.  I don't remember when the change happened.

dgrigg123, 2017-12-10:
Thanks.  I tried it today and works fine now.  However, since I've  been flying without it for a while I find it a bit strange.  It seems to effect my roll rate a bit AND it makes some odd wind-down sounds from the motors after rolls, as if there is some deceleration rate of the motors being limited.  I don't know.  Maybe just need to get used to it.
Thanks again.

etracer65, 2017-12-10:
Yes, it will slow the roll rate as it has to coordinate yaw with the roll.  To yaw one of the motors generating the roll has to be reduced in power and the opposite of the other increased.  This is then fighting against the roll.  It does feel very strange and why most people don't use it - particularly if you've been flying for a while.  Try flying LOS and do a roll - that looks really weird!

If this is resolved go ahead and close the issue.

closed 2017-12-11 by dgrigg123