betaflight/betaflight #4732: Gyro not calibrating or similiar

airexes, 2017-12-10:
Hello, I am having some very frustration issues with my 2s 95mm micro quad. I have betaflight 3.2.2 and an omnibusf4sd with an mpu6000, d shot 600 and 8k/8k. I have tried a few different firmware versions and every setting imaginable as well as 3 different identical boards and esc's. I tried manually redoing all settings after a complete board wipe as well as config dumps. I have tried soft mounting with 0 change and my props are balanced. I always let the gyro calibrate after plugging the battery in. I have also tried manually calibrating with stick commands.

A while ago with an older identical board everything was fine.

Sometimes when I change just one pid value it becomes almost unflyable. I try to put the value back but it doesn't help. I have to reduce the pids down to 20 for it to fly without an extreme amount of oscillation. At some point I can raise them again and all is okay, the loop continues again at some point with changing one value slightly. I have tried basically every combination of pids. 

When I arm with decent pids sometimes it will have so much noise in the motors they get hot in a few seconds and the craft also nose dives forwards. This will happen over and over and then at some point I get frustrated. I come back later and it arms and flies smooth again. I make sure to let the gyro calibrate whenever I plug the battery in. These happens right out of the gate even on stock betaflight pids. I have tried manually changing all settings as well as dumping cli config in to no effect.

I have tried disabling the dynamic filter, running every possible combination of filters and pt1/biquad filter. Honestly no settings changed even on my old board that didn't seem to have issues ever tamed the propwash. 

On my old board I had all the filters removed running dynamic and had the p and i in the 70's even with the d gain 30-35, no issues. Now I'm lucky to get 40 p and i with 18 d gain. Going over 18 usually leads to a huge amount of noise out of nowhere, but sometimes not. Sometimes I can adjust it to 27 before noise happens.

The pids are low, just now I raised p to 55 and p pitch to 70 and it flew fine. Then without changing the settings it just started oscillating like crazy again.

I wrote this up yesterday, at the time there was so much noise in the motors and it just fell on it's face. Came back this morning and hooked a new battery up, smooth as butter. The cycle doesn't stop.

2017-12-09 @ 20:47:57 -- Running - OS: Windows, Chrome: 62.0.3202.94, Configurator: 3.2.2
2017-12-09 @ 20:48:03 -- Serial port successfully opened with ID: 1
2017-12-09 @ 20:48:03 -- MultiWii API version: 1.36.0
2017-12-09 @ 20:48:03 -- Flight controller info, identifier: BTFL, version: 3.2.2
2017-12-09 @ 20:48:03 -- Running firmware released on: Nov 11 2017 00:39:14
2017-12-09 @ 20:48:03 -- Board: OBSD, version: 0
2017-12-09 @ 20:48:03 -- Unique device ID: 0x3f004a3436510b38363236
2017-12-09 @ 20:48:03 -- Craft name:
2017-12-09 @ 20:48:13 -- CLI mode detected
Setup
Ports
Configuration
Power & Battery
PID Tuning
Receiver
Modes
Motors
OSD
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.2.2 Nov 11 2017 / 00:39:14 (77782e602) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1425 2100
aux 1 28 1 1525 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 100
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set serialrx_provider = SBUS
set dshot_idle_value = 380
set motor_pwm_protocol = DSHOT600
set align_board_roll = 180
set deadband = 1
set pid_process_denom = 1
set osd_rssi_pos = 2445
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2455
set osd_flymode_pos = 333
set osd_throttle_pos = 2433
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set anti_gravity_gain = 2000
set setpoint_relax_ratio = 37
set dterm_setpoint_weight = 36
set p_pitch = 57
set d_pitch = 26
set p_roll = 50
set i_roll = 45
set d_roll = 22

# rateprofile
rateprofile 0

set rc_rate = 75
set rc_rate_yaw = 150
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 60
set tpa_rate = 35

#

mikeller, 2017-12-10:
Sounds a lot like you've got a board with a bad gyro chip - get a warranty replacement.

airexes, 2017-12-10:
No I tried 3 different boards. The problem intermittent. It's 100% perfect and great, next it's as terrible as can be. It is like it randomly turns every filter off including the dynamic one.

etracer65, 2017-12-10:
I'm noting that you seem to have the dterm notch disabled.  Unless you have a really clean and vibration free quad (and it sounds like you don't) you really should turn the dterm notch back on.  Also you are using PT1 for the dterm lowpass - again not recommended for noisy setups.  Try changing back to BIQUAD and re-enabling the dterm notch.

airexes, 2017-12-10:
My setup isn't noisy. I wrote 3 paragraphs explaining how it flies completely fine with no notch filters and then all of the sudden won't even hover without wobbling itself to death. 

Currently I have a perfect tune on it with no notch filters at all and it is flying perfect, I almost completely got rid of the prop wash. I will probably come back later today, plug a battery in and it won't even hover. The motors get too hot to touch in seconds when this happens.

It's not my tune. It's not the hardware. It's like the gyro won't calibrate when this happens.

etracer65, 2017-12-10:
Without the dterm notch and runnint PT1 lowpass all it takes is for a prop to get slightly out of balance and that can push it over the edge.  The effects are exactly what you describe happening - primarily massive dterm oscillations.

Start but putting all the filtering back to defaults - both gyro notches and dterm notch/type.  If the problems persist then maybe you have a bad/failing gyro.

If the problems clear up, then start retuning your filters.  Joshua Bardwell has a good video on how to do this safely and methodically.

I suspect you've just gone overboard on removing filtering.  Very few quads can get away with no gyro and dterm notches and the type set to PT1.

airexes, 2017-12-10:
You aren't understanding. It does this on stock pids and filters too. It flies great for a while. Next thing you know without changing anything it oscillates like crazy. That's with all the filters on, stock settings with lowered d gains.

It's like it's either randomly removing filters or not calibrating the gyro properly.

This doesn't just happen once. Either it flies great or it doesnt. When it doesn't if I come back hours later it sometimes just works again. I Un plug the battery and re plug it 50 times and it still won't work. Hours later it magically flies great again. No settings changed.

etracer65, 2017-12-10:
Well perhaps you can post some blackbox logs (with the default filter settings) showing examples of when it is and is not happening?  Use "debug_mode = notch" in the CLI and set the blackbox logging rate to 2KHz. I suspect in the cases where it's happening we'll see large amounts of gyro noise.

Why it comes and goes is unknown.  The flight controller has no "memory" of events after you cycle power so it must be something external.  Possibly a loose connection, motor screw touching a winding, failing or noisy BEC, harmonic oscillation in the frame, etc.  All of which would be affected by external conditions such as temperature and/or vibration.  You said you've replaced the FC so it's time to look at other components.

If the gyro failed to calibrate the effect would be a "Tasmanian Devil" as soon as the PID controller became active.  Been there and done that!  The quad wouldn't fly at all without gyro calibration.

airexes, 2017-12-10:
I have tried everything physical I could think of. It hasn't happened for a few days but remember I said I would land, change a value very slightly in osd and then it would oscillate badly. Changing the value back did not help it would continue to behave poorly.

jflyper, 2017-12-10:
@airexes You are not following the instructions given by the pre-formatted portion of the issues template. What is your FC?

jflyper, 2017-12-10:
@airexes I would also like to see sensors's tab with gyro and acc turned on, resting still on table top, when the craft is not arming.

fftunes, 2017-12-12:
Just a note, i once had something similar due to a slipping magnet... When it was in the right place, everything was fine, when it slipped, it produced mad oscillations. Sometimes i could provoque the slipping by hard acceleration/braking.